introlab/rtabmap #765: RTAB-Map not compiling after updating librealsense

KierenSama2, 2021-08-19:
I built rtabmap from source on neotic 20.04 and it was working fine. I recently updated librealsense and have been having issues since then. When I try run $ rtabmap I get the following output.
`rtabmap: error while loading shared libraries: librealsense2.so.2.41: cannot open shared object file: No such file or directory`
When I locate the missing library I get:
`locate librealsense2.so
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense2.so
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense2.so.2.48
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense2.so.2.48.0
/usr/lib/x86_64-linux-gnu/librealsense2.so
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.49
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.49.0
`
How can I fix this issue?

KierenSama2, 2021-08-19:
I should also add, that when I run '$ realsense-viewer it runs v2.49.0

KierenSama2, 2021-08-19:
Issue was resolved.
`$ cd rtabmap/build`
`$ cmake ..  [<---double dots included]`
`$ make`
`$ sudo make install`

I think this updated all of the links. And then after this, catkin_make in  /catkin_ws or wherever rtabmap_ros is

closed 2021-08-19 by KierenSama2

matlabbe, 2021-09-03:
If you update realsense version (or any other rtabmap's dependencies) after building rtabmap with an older one, the safest approach is to cleanup the build directory of rtabmap to make sure it rebuilds and links everything with the right updated libraries.