introlab/rtabmap #766: Error build with Loam_Velodyne

jacklu333333, 2021-08-23:
Hi, I am want to rebuild rtabmap with loam_velodyne to take the advantage of loam_velodyne algorithm. However, when I tried to build the package, the following error
```
/home/jack/rtabmap/corelib/src/odometry/OdometryLOAM.cpp:286:169: error: no matching function for call to ‘rtabmap::LaserScan::backwardCompatibility(rtabmap::LaserScan, int, float, rtabmap::Transform)’
      info->localScanMap = LaserScan::backwardCompatibility(util3d::laserScanFromPointCloud(out), 0, data.laserScanRaw().rangeMax(), data.laserScanRaw().localTransform());
```
Is there any way I can resolve this problem?

Thanks in advance.

jacklu333333, 2021-08-23:
Hi,
It seems there are similar questions and answers on rtabmap forum. But I cannot access them.
I got `ERR_NAME_NOT_RESOLVED` on my web page. Is it still possible to access the rtabmap forum?

[build error ubuntu 20.04 ROS Noetic (LOAM)](http://official-rtab-map-forum.67519.x6.nabble.com/build-error-ubuntu-20-04-ROS-Noetic-LOAM-td8187.html)

matlabbe, 2021-08-24:
I updated your link, the server has changed address, unfortunatly. See http://official-rtab-map-forum.206.s1.nabble.com/Forum-address-has-changed-how-fixing-old-reference-links-not-working-anymore-td8425.html if you come with another broken link.