introlab/rtabmap #769: PCL requirements?

parkerlreed, 2021-09-05:
Arch Linux

pcl 1.12 and pcl master are both not working for compilation currently.

1.12
```
[ 22%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_correspondences.cpp.o
In file included from /home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/pcl18/surface/texture_mapping.h:473,
                 from /home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/util3d_surface.cpp:48:
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/pcl18/surface/impl/texture_mapping.hpp: In member function ‘bool pcl::TextureMapping<PointInT>::textureMeshwithMultipleCameras2(pcl::TextureMesh&, const CameraVector&, const rtabmap::ProgressState*, std::vector<std::map<int, pcl::PointXY> >*)’:
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/pcl18/surface/impl/texture_mapping.hpp:1142:59: error: ‘getAngle3D’ is not a member of ‘pcl’
 1142 |                                         angleToCam = pcl::getAngle3D(Eigen::Vector4f(normal3D[0], normal3D[1], normal3D[2], 0.0f), Eigen::Vector4f(0.0f,0.0f,-1.0f,0.0f));
      |                                                           ^~~~~~~~~~
[ 22%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_motion_estimation.cpp.o
[ 23%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/SensorData.cpp.o
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:711: corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:790: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

master
```
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o
In file included from /home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/util3d_surface.h:475,
                 from /home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/Memory.cpp:51:
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp: In function ‘void rtabmap::util3d::denseMeshPostProcessing(pcl::PolygonMeshPtr&, float, int, const typename pcl::PointCloud<PointT>::Ptr&, float, bool, bool, int, rtabmap::ProgressState*)’:
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:108:22: error: ‘fromPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  108 |                 pcl::fromPCLPointCloud2(mesh->cloud, *coloredCloud);
      |                      ^~~~~~~~~~~~~~~~~~
      |                      PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:153:22: error: ‘toPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  153 |                 pcl::toPCLPointCloud2(*coloredCloud, mesh->cloud);
      |                      ^~~~~~~~~~~~~~~~
      |                      PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:247:30: error: ‘fromPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  247 |                         pcl::fromPCLPointCloud2(mesh->cloud, *cloud);
      |                              ^~~~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:285:30: error: ‘toPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  285 |                         pcl::toPCLPointCloud2 (*cloud, mesh->cloud);
      |                              ^~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:290:30: error: ‘fromPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  290 |                         pcl::fromPCLPointCloud2(mesh->cloud, *cloud);
      |                              ^~~~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:315:30: error: ‘toPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  315 |                         pcl::toPCLPointCloud2 (*cloud, mesh->cloud);
      |                              ^~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
[  8%] Linking CXX executable /home/parker/build/ros-noetic-rtabmap/src/rtabmap/bin/rtabmap-extractObject
[  8%] Built target extractObject
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:123: corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:790: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

CMake output
```
CMake Deprecation Warning at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
[... 508 lines cut ...]
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (VINS-Fusion not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
```

parkerlreed, 2021-09-05:
I realized I mistakenly built 1.11. Compiling and retesting with 1.12 now.

parkerlreed, 2021-09-05:
Same error as master on 1.12

```
[  8%] Linking CXX executable /home/parker/build/ros-noetic-rtabmap/src/rtabmap/bin/rtabmap-extractObject
In file included from /home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/util3d_surface.h:475,
                 from /home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/Memory.cpp:51:
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp: In function ‘void rtabmap::util3d::denseMeshPostProcessing(pcl::PolygonMeshPtr&, float, int, const typename pcl::PointCloud<PointT>::Ptr&, float, bool, bool, int, rtabmap::ProgressState*)’:
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:108:22: error: ‘fromPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  108 |                 pcl::fromPCLPointCloud2(mesh->cloud, *coloredCloud);
      |                      ^~~~~~~~~~~~~~~~~~
      |                      PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:153:22: error: ‘toPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  153 |                 pcl::toPCLPointCloud2(*coloredCloud, mesh->cloud);
      |                      ^~~~~~~~~~~~~~~~
      |                      PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:247:30: error: ‘fromPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  247 |                         pcl::fromPCLPointCloud2(mesh->cloud, *cloud);
      |                              ^~~~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:285:30: error: ‘toPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  285 |                         pcl::toPCLPointCloud2 (*cloud, mesh->cloud);
      |                              ^~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:290:30: error: ‘fromPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  290 |                         pcl::fromPCLPointCloud2(mesh->cloud, *cloud);
      |                              ^~~~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
/home/parker/build/ros-noetic-rtabmap/src/rtabmap/corelib/src/../include/rtabmap/core/impl/util3d_surface.hpp:315:30: error: ‘toPCLPointCloud2’ is not a member of ‘pcl’; did you mean ‘PCLPointCloud2’?
  315 |                         pcl::toPCLPointCloud2 (*cloud, mesh->cloud);
      |                              ^~~~~~~~~~~~~~~~
      |                              PCLPointCloud2
[  8%] Built target extractObject
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:123: corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

matlabbe, 2021-09-06:
Cannot reproduce the problem with PCL 1.12 and [pcl master](https://github.com/PointCloudLibrary/pcl/commit/ce8d85d1237bd0e8ac2d94446a43159adf00c818):
```
./rtabmap --version 
RTAB-Map:              0.20.13
PCL:                    1.12.0
With VTK:                9.0.1
OpenCV:                  4.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With Madgwick:            true
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:               true
With OpenNI2:             true
With Freenect:            true
With Freenect2:           true
With K4W2:               false
With K4A:                 true
With DC1394:              true
With FlyCapture2:        false
With ZED:                 true
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:             true
With libpointmatcher:     true
With CCCoreLib:           true
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```
There is maybe a problem with the include paths, because `pcl::fromPCLPointCloud2` and `pcl::toPCLPointCloud2` do really exist in the include path here:
https://github.com/introlab/rtabmap/blob/c43118cde89b6e1bbb3a7ade6875fc7ee488f9a1/corelib/include/rtabmap/core/impl/util3d_surface.hpp#L12
conversions.h: https://github.com/PointCloudLibrary/pcl/blob/master/common/include/pcl/conversions.h#L166-L347

parkerlreed, 2021-09-06:
Thank you for pointing out that include line. I HAD looked there earlier and it wasn't there. Realized the script I used pulled an old tag from rtabmap-releases repo, ugh. ACTUALLY building master now.

parkerlreed, 2021-09-06:
Success! Thanks again.

```
[parker@wolfcola ros-noetic-rtabmap]$ rtabmap --version
RTAB-Map:              0.20.13
PCL:                    1.12.0
With VTK:                8.2.0
OpenCV:                  4.5.3
With OpenCV xfeatures2d:  true
With OpenCV nonfree:      true
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With Madgwick:            true
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:            true
With Freenect2:           true
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

closed 2021-09-06 by parkerlreed