introlab/rtabmap #770: Help for a Launch file of 3D laser and D455i

farhangnaderi, 2021-09-19:
I am trying to run rtabmap_ros and went through demo hector mapping launch file in addition to Velodyne Puck example but got confused competely and none give my any output on Rviz. 

My setup consists of a 3D lidar with 75 meters range publishing to PointCloud2 and a Realsense D455i. Currently, I can visualize my cloud from the 3D lidar only on Rviz with laser_link frame but not with rtabmap. 

Any help to guide me for a launch file to both mapping from the 3D lidar and/or a combination of lidar and my D455i is highly appreciated.

Thanks in advance.

matlabbe, 2021-09-20:
The [demo hector mapping example](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#A2D_laser_only) is for 2D lidar. For 3D lidar, look at the husky launch file: 
https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_husky.launch.

Is it on a robot or hand-held setup? Do you have external wheel odometry available? Do you want to do lidar odometry or RGB-D odometry?

For icp_odometry examples, rtabmap should be built with libpointmatcher.

farhangnaderi, 2021-09-20:
Everything is set up on the robot. I still do not have the odometry node and am planning to rely on the Lidar for odometry now. 
The other problem is I have to build from the source for libpointmatcher since I am using a Jetson device and is not passing successfully.

matlabbe, 2021-09-20:
Is it libpointmatcher or rtabmap that has build errors? What are they?

farhangnaderi, 2021-09-20:
Yes I had build errors which I donhave the logs now. Back to the office this week I will share them. Thanks for your quick response by the way.

farhangnaderi, 2021-09-27:
I used husky launch file as edited below:
https://gist.github.com/farhangnaderi/43e14e70244fe245ba1957c878a8b416

But now I am receiving this output:

```
[ WARN] [1632728343.780819047]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=5).
/rtabmap/rtabmap subscribed to (approx sync):
   /odometry/filtered \
   /lslidar_point_cloud

I have changed the queue size as well bu no change. Also tried icp and again same. My 3D lidar is publishing at 10hz. 
```

matlabbe, 2021-09-28:
I launch your launch file, I don't have the same topics:
```
roslaunch 3d_lidar.launch lidar3d:=true

[ WARN] [1632866682.070991077]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/odom \
   /lslidar_point_cloud \
   /rtabmap/odom_info
```
When topics are not received, try what is explained in the warning, do `rostopic hz` with the topics to see if they are published. In your case, is `/odometry/filtered` also published? Which node is publishing it?

farhangnaderi, 2021-09-28:
On the second one (lslidar) I am sure I have published data. But I have nothing on odom. Even icp odom didnt change scenario, again same warning. Maybe I am missing a step in docs? No idea.

matlabbe, 2021-09-30:
From the original husky demo:
https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_husky.launch

The would be something similar to 10:
```
roslaunch rtabmap_ros demo_husky.launch lidar3d:=true slam2d:=false camera:=false icp_odometry:=true rtabmapviz:=true
```
but you need to remove https://github.com/introlab/rtabmap_ros/blob/a5fe8df170c990d630fc14895f0c996968d3f14e/launch/demo/demo_husky.launch#L113 as you don't have any other odometry.

You may also try [test_velodyne.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_velodyne.launch) launch example (remove [this](https://github.com/introlab/rtabmap_ros/blob/a5fe8df170c990d630fc14895f0c996968d3f14e/launch/tests/test_velodyne.launch#L33-L37)):
```
roslaunch rtabmap_ros test_velodyne.launch scan_topic:=/lslidar_point_cloud frame_id:=yuvy
```

farhangnaderi, 2022-01-25:
I was missing for a while and am back. 
I will try everything from the begining. Will close this issue and will open another after a progress or re-open this again.

closed 2022-01-25 by farhangnaderi

Danendra10, 2024-04-05:
Any update here? trying to integrate rtabmap with velodyne lidar, odom, and without gps

matlabbe, 2024-04-06:
ros1 or ros2? For ros1, what you described seems that config:
https://github.com/introlab/rtabmap_ros/blob/665735663ef7bf029c11c599c06a1fd727299e0c/rtabmap_demos/launch/demo_husky.launch#L16-L17