introlab/rtabmap #772: Compilation Error Ubuntu 18.04

David9696, 2021-09-27:
Hi,

I got the following error when I make the package(latest):

[ 80%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `OdomEstimationClass::initMapWithPoints(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> > const&, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> > const&)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `OdomEstimationClass::init(lidar::Lidar, double)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `lidar::Lidar::setMinDistance(double)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `LaserProcessingClass::LaserProcessingClass()'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `LaserProcessingClass::featureExtraction(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> > const&, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> >&, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> >&)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `LaserProcessingClass::init(lidar::Lidar)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `lidar::Lidar::Lidar()'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `lidar::Lidar::setVerticalAngle(double)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `OdomEstimationClass::OdomEstimationClass()'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `lidar::Lidar::setLines(double)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `lidar::Lidar::setMaxDistance(double)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `OdomEstimationClass::updatePointsToMap(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> > const&, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> > const&)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `lidar::Lidar::setScanPeriod(double)'
../../../bin/librtabmap_core.so.0.20.14: undefined reference to `OdomEstimationClass::getMap(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> >&)'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:746: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:1753: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

Is it something missing?

Thanks.

Regards,

David

matlabbe, 2021-09-27:
Do you have FLOAM installed? To build with FLOAM, you should use this patched version: http://official-rtab-map-forum.206.s1.nabble.com/icp-odometry-with-LOAM-crash-tp8261p8563.html

To build without FLOAM, do `cmake -DWITH_FLOAM=OFF ..`.

matlabbe, 2021-09-28:
I didn't test on Melodic, but I guess it could work. I think the patch https://gist.github.com/matlabbe/d8ab56e69146c18afbd1a3b05444c649 is compatible with Melodic. You should use original FLOAM repo though.

anath93, 2023-08-01:
@matlabbe Will this patch will still work for ROS2 Foxy ?

matlabbe, 2023-08-05:
There is no known patch for ROS2. If you only need floam library in ROS2, I would remove everything ROS1, and just keep the minimal library without any ROS.

anath93, 2023-08-05:
@matlabbe Thank you.