introlab/rtabmap #779: Power saving throttle

roppppp, 2021-10-31:
Hi there,
while scanning I noticed a significant performance drop as I disconnected my power cord.
Is it possible to disable power savings for this program or is this done by Windows ?
Tried to disable all power savings in bios and windows, but still no luck.
There should also be a warning in the console, I only noticed this issue after lots of headaches and scans.
The only thing I can think of is to try linux.

Cheers

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel L515  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 10 Pro 21H1 x64  | 
| Platform                                 | PC USB 3.2  |
| Realsense SDK Version                            |  2.49.0                          | 
| RTabMap                           |  0.20.8 CUDA                         | 
| Source input rate | 0 |
| Segment			|  Mapping                        | 
| Laptop | Thinkpad P51 20HH0018GE 16GB |
| Btw | RTabMap is awesome |

roppppp, 2021-11-01:
Found out that it was caused by the CPU throttling, this seems to be a model specific problem,
https://forums.lenovo.com/t5/ThinkPad-P-and-W-Series-Mobile-Workstations/P51-Heavy-Throttling-on-Battery/m-p/3693399?page=8
The Intel CPU starts throttling with BD PROCHOT, i disabled it with ThrottleStop from techpowerup.com as described in the lenovo forum.
Even if this feature is used to keep the CPU from overheating and safe battery consumption , i have 0 problems if I run it in max mode, so its a lenovo bug to me. If its possible it would be useful for others to throw this message into the console if BD PROCHOT starts activating -> it throttles the CPU from 3800MHz to 800MHz in my case.

matlabbe, 2021-11-04:
I've seem on some Windows computers that implementation of OpenMP by default on windows keeps the threads alive even if they are not used, causing a 100% CPU usage on all cores even by just showing the camera frames without any processing. Without recompiling RTAB-Map on Windows with OpenMP disabled (-DWITH_OPENMP=OFF), you can add [OMP_WAIT_POLICY](https://www.openmp.org/spec-html/5.0/openmpse55.html) to system [environment variables](https://docs.oracle.com/en/database/oracle/machine-learning/oml4r/1.5.1/oread/creating-and-modifying-environment-variables-on-windows.html#GUID-DD6F9982-60D5-48F6-8270-A27EC53807D0) to make OpenMP not doing an active waiting:
```
OMP_WAIT_POLICY=passive
```

closed 2021-11-04 by matlabbe