introlab/rtabmap #780: Unable to build the foxy-devel branch

ericdepotter, 2021-11-02:
I'm trying to get this repository up and running in a ROS2 foxy environment. However, when trying to build the foxy-devel branch, I get the following error:

```
--- stderr: rtabmap In file included from /usr/local/include/depthai/device/CalibrationHandler.hpp:5, 
from /usr/local/include/depthai/depthai.hpp:9, 
from /home/eric/ROS/dev_ws/src/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraDepthAI.h:40,
from /home/eric/ROS/dev_ws/src/rtabmap/corelib/src/camera/CameraDepthAI.cpp:28: /usr/local/include/depthai-shared/common/CameraBoardSocket.hpp:3:10:
fatal error: nlohmann/json.hpp: No such file or directory 
  3 | #include <nlohmann/json.hpp> 
    |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I followed these installation steps:

```bash
git clone --branch foxy-devel https://github.com/introlab/rtabmap.git src/rtabmap
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro foxy
export MAKEFLAGS="-j6"
colcon build --symlink-install
```

matlabbe, 2021-11-04:
Try master branch (there are some fixes added with latest depthai version [here](https://github.com/introlab/rtabmap/commit/32bf0f9d61c8aacf611c5c9c6b6349a15c48dc62)), though it seems DepthAI related issue. If you don't need depthai built-in with rtabmap, disable it with:
```
colcon build --symlink-install --cmake-args -DWITH_DEPTHAI=OFF
```

ericdepotter, 2021-11-07:
I tried with the master branch initially, but this gave errors with the ros2 branch of rtabmap_ros (see [here](https://github.com/introlab/rtabmap_ros/issues/670)). However there they said it should work with master, so I decided to try again.

Hence, I pulled the latest changes from master, but now I cannot build it anymore. I get the following error:
```--- stderr: rtabmap                               
/usr/bin/ld: /home/eric/ROS/dev_ws/src/rtabmap/bin/librtabmap_core.so.0.20.15: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/build.make:262: /home/eric/ROS/dev_ws/src/rtabmap/bin/rtabmap-stereoEval] Error 1
make[1]: *** [CMakeFiles/Makefile2:1131: tools/StereoEval/CMakeFiles/stereoEval.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
The `ros-foxy-octomap` package is installed.

matlabbe, 2021-11-14:
Is `liboctomap-dev` also installed? I see that `liboctomap-dev` version is 1.9.3 while `ros-foxy-octomap` is 1.9.7. Note sure if it could be the problem. On my machine, I have only `liboctomap-dev`.

ericdepotter, 2021-11-24:
```liboctomap-dev``` was already installed with version 1.9.3. Removing ```ros-foxy-octomap``` fixed the build issue. Thanks for the help in figuring out the issue.

closed 2021-11-24 by ericdepotter