introlab/rtabmap #781: New rtabmap bloom release for ROS.

Timple, 2021-11-08:
`rtabmap_ros` now depends on `rtabmap 0.20.15`.

This means the whole of `rtabmap` needs to be compiled in order to use the latest `rtabmap_ros`.

Would you mind rerunning a bloom release for (at least) galactic?

matlabbe, 2021-11-08:
I did new bloom releases:
Foxy: https://github.com/ros/rosdistro/pull/31163
Galactic: https://github.com/ros/rosdistro/pull/31164
Rolling: https://github.com/ros/rosdistro/pull/31165

Timple, 2021-11-09:
Thank you! That was very responsive :slightly_smiling_face: 

Would it make sense to release `rtabmap_ros` as well so users never go out-of-sync on the versions? Or am I being too forward here?

matlabbe, 2021-11-14:
rtabmap_ros for foxy and galactic pull requests have been made on rosdistro, they would appear in next ros2 release.