introlab/rtabmap #783: Problem with OKVIS Odometry Strategy

Sara-Esam, 2021-11-12:
Hi, 

I was working on a project testing OKVIS Odometry with RTAB-MAP, and I came across this problem:

matlabbe, 2021-11-15:
How did you build OKVIS? Make sure to remove all `-march=native` from cmake files: https://github.com/ethz-asl/okvis/search?q=march%3Dnative

Sara-Esam, 2021-11-15:
Thanks for your fast response! 

Indeed the flag `-march=native` was the problem. 

However, I am still facing another issue: 

[image]
[image]

Is it due to Ceres version? 
OKVIS is using 1.9.0 
But in rtabmap I use 1.14.0 because otherwise it gives this building error: 

../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/Recovery/CMakeFiles/recovery.dir/build.make:518: recipe for target '../bin/rtabmap-recovery' failed
make[2]: *** [../bin/rtabmap-recovery] Error 1
CMakeFiles/Makefile2:1219: recipe for target 'tools/Recovery/CMakeFiles/recovery.dir/all' failed
make[1]: *** [tools/Recovery/CMakeFiles/recovery.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/StereoEval/CMakeFiles/stereoEval.dir/build.make:518: recipe for target '../bin/rtabmap-stereoEval' failed
make[2]: *** [../bin/rtabmap-stereoEval] Error 1
CMakeFiles/Makefile2:1107: recipe for target 'tools/StereoEval/CMakeFiles/stereoEval.dir/all' failed
make[1]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/all] Error 2
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/build.make:518: recipe for target '../bin/rtabmap-rgbd_dataset' failed
make[2]: *** [../bin/rtabmap-rgbd_dataset] Error 1
CMakeFiles/Makefile2:1163: recipe for target 'tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/all' failed
make[1]: *** [tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/all] Error 2
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/Camera/CMakeFiles/camera.dir/build.make:518: recipe for target '../bin/rtabmap-camera' failed
make[2]: *** [../bin/rtabmap-camera] Error 1
CMakeFiles/Makefile2:1079: recipe for target 'tools/Camera/CMakeFiles/camera.dir/all' failed
make[1]: *** [tools/Camera/CMakeFiles/camera.dir/all] Error 2
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/EurocDataset/CMakeFiles/euroc_dataset.dir/build.make:519: recipe for target '../bin/rtabmap-euroc_dataset' failed
make[2]: *** [../bin/rtabmap-euroc_dataset] Error 1
CMakeFiles/Makefile2:1191: recipe for target 'tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all' failed
make[1]: *** [tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all] Error 2
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/KittiDataset/CMakeFiles/kitti_dataset.dir/build.make:518: recipe for target '../bin/rtabmap-kitti_dataset' failed
make[2]: *** [../bin/rtabmap-kitti_dataset] Error 1
CMakeFiles/Makefile2:1135: recipe for target 'tools/KittiDataset/CMakeFiles/kitti_dataset.dir/all' failed
make[1]: *** [tools/KittiDataset/CMakeFiles/kitti_dataset.dir/all] Error 2
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
[... cut ...]
../../../bin/librtabmap_core.so.0.20.11: undefined reference to `vtable for ceres::EigenQuaternionParameterization'
collect2: error: ld returned 1 exit status
tools/Reprocess/CMakeFiles/reprocess.dir/build.make:520: recipe for target '../bin/rtabmap-reprocess' failed
make[2]: *** [../bin/rtabmap-reprocess] Error 1
CMakeFiles/Makefile2:1247: recipe for target 'tools/Reprocess/CMakeFiles/reprocess.dir/all' failed
make[1]: *** [tools/Reprocess/CMakeFiles/reprocess.dir/all] Error 2
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2

matlabbe, 2021-11-15:
Rtabmap should not take system ceres with okvis. It should take ceres version (1.9.0) built inside okvis. 
https://github.com/introlab/rtabmap/blob/7af2a27e893b8c3ec54121c551c3d03f7304a769/CMakeLists.txt#L658

If you built okvis as-is, you may get ceres linker errors on rtabmap side. See this patch: https://gist.github.com/matlabbe/383be55b5cb682fea217d45ef9a37ef8

Okvis was built like this in the test [dockerfile](https://github.com/introlab/rtabmap/blob/7af2a27e893b8c3ec54121c551c3d03f7304a769/docker/jfr2018/latest/Dockerfile#L145):
```bash
# Clone and apply patch
$ git clone https://github.com/ethz-asl/okvis.git 
$ cd okvis && git checkout 1dce9129f22dd4d21d944788cd9da3a4341586aa 
$ wget https://gist.githubusercontent.com/matlabbe/383be55b5cb682fea217d45ef9a37ef8/raw/bd4f6886eeba3800f0e5171e77abc5dd6dc8bfd9/okvis_1dce912_marchnative_disabled.patch
$ git apply --ignore-space-change --ignore-whitespace okvis_1dce912_marchnative_disabled.patch

# Build and install
$ mkdir build 
$ cd build 
$ cmake ..
$ make -j6
$ sudo make install 
$ sudo mkdir -p /usr/local/lib/cmake 
$ sudo mv /usr/local/lib/CMake/* /usr/local/lib/cmake/.
```

Sara-Esam, 2021-11-15:
Thank you for clarifying

I built OKVIS the same way as you posted.
Ceres 1.9.0 is being used, but rtabmap build gives eigen errors when building OptimizeCeres.cpp: 

` error: no type named ‘Literal’ in ‘struct Eigen::NumTraits<ceres::Jet<double, 6> >’`

`error: no type named ‘Literal’ in ‘struct Eigen::NumTraits<ceres::Jet<double, 14> >’
 struct promote_scalar_arg<S,T,false> : promote_scalar_arg_unsupported<S,T,typename NumTraits<S>::Literal> {};`

This is the full build log: 

`[  4%] Built target rtabmap_utilite
[  4%] Built target imagesJoiner
[  5%] Built target res_tool
[  6%] Built target extractObject
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/StereoDense.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/MarkerDetector.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Stereo.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/StereoCameraModel.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/stereo/StereoBM.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OccupancyGrid.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/stereo/StereoSGBM.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerCeres.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/GainCompensator.cpp.o
[  9%] Building C object corelib/src/CMakeFiles/rtabmap_core.dir/rtflann/ext/lz4.c.o
[  9%] Building C object corelib/src/CMakeFiles/rtabmap_core.dir/rtflann/ext/lz4hc.c.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/FlannIndex.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/discrete_depth_distortion_model_helpers.cpp.o
In file included from /home/xliu/rtabmap/corelib/src/rtflann/util/params.h:33:0,
                 from /home/xliu/rtabmap/corelib/src/rtflann/flann.hpp:42,
                 from /home/xliu/rtabmap/corelib/src/FlannIndex.cpp:31:
/home/xliu/rtabmap/corelib/src/rtflann/util/any.h: In member function ‘virtual void rtflann::anyimpl::big_any_policy<T>::static_delete(void**)’:
/home/xliu/rtabmap/corelib/src/rtflann/util/any.h:81:9: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
         if (* x) delete (* reinterpret_cast<T**>(x)); *x = NULL;
         ^~
/home/xliu/rtabmap/corelib/src/rtflann/util/any.h:81:55: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
         if (* x) delete (* reinterpret_cast<T**>(x)); *x = NULL;
                                                       ^
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/frame_projector.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/discrete_depth_distortion_model.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/slam_calibrator.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/ORBextractor.cc.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/solvepnp.cpp.o
In file included from /usr/include/eigen3/Eigen/Core:348:0,
                 from /home/xliu/rtabmap/corelib/src/../include/rtabmap/core/Transform.h:35,
                 from /home/xliu/rtabmap/corelib/src/../include/rtabmap/core/Link.h:33,
                 from /home/xliu/rtabmap/corelib/src/../include/rtabmap/core/Graph.h:36,
                 from /home/xliu/rtabmap/corelib/src/optimizer/OptimizerCeres.cpp:27:
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h: In instantiation of ‘struct Eigen::internal::promote_scalar_arg<ceres::Jet<double, 14>, Eigen::Quaternion<ceres::Jet<double, 14>, 0>, false>’:
/usr/include/eigen3/Eigen/src/plugins/CommonCwiseBinaryOps.h:50:1:   required by substitution of ‘template<class T> typename Eigen::internal::enable_if<true, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<typename Eigen::internal::promote_scalar_arg<ceres::Jet<double, 14>, T, Eigen::internal::has_ReturnType<Eigen::ScalarBinaryOpTraits<T, ceres::Jet<double, 14>, Eigen::internal::scalar_product_op<T, ceres::Jet<double, 14> > > >::value>::type, ceres::Jet<double, 14> >, const typename Eigen::internal::plain_constant_type<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<ceres::Jet<double, 14>, ceres::Jet<double, 14> >, const Eigen::Map<const Eigen::Matrix<ceres::Jet<double, 14>, 3, 1, 0, 3, 1>, 0, Eigen::Stride<0, 0> >, const Eigen::Map<const Eigen::Matrix<ceres::Jet<double, 14>, 3, 1, 0, 3, 1>, 0, Eigen::Stride<0, 0> > >, typename Eigen::internal::promote_scalar_arg<ceres::Jet<double, 14>, T, Eigen::internal::has_ReturnType<Eigen::ScalarBinaryOpTraits<T, ceres::Jet<double, 14>, Eigen::internal::scalar_product_op<T, ceres::Jet<double, 14> > > >::value>::type>::type, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<ceres::Jet<double, 14>, ceres::Jet<double, 14> >, const Eigen::Map<const Eigen::Matrix<ceres::Jet<double, 14>, 3, 1, 0, 3, 1>, 0, Eigen::Stride<0, 0> >, const Eigen::Map<const Eigen::Matrix<ceres::Jet<double, 14>, 3, 1, 0, 3, 1>, 0, Eigen::Stride<0, 0> > > > >::type Eigen::operator*(const T&, const StorageBaseType&) [with T = Eigen::Quaternion<ceres::Jet<double, 14>, 0>]’
/home/xliu/rtabmap/corelib/src/optimizer/ceres/pose_graph_3d/pose_graph_3d_error_term.h:92:57:   required from ‘bool ceres::examples::PoseGraph3dErrorTerm::operator()(const T*, const T*, const T*, const T*, T*) const  [... cut ...]
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:903: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerCeres.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerCeres.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:912: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2`

matlabbe, 2021-11-15:
As ceres is found by OKVIS, rtabmap was building OptimizerCeres.cpp, but it seems 1.9.0 is causing some building errors. I updated in the commit above that with WITH_CERES=OFF, OptimizerCeres.cpp won't be built:
```bash
git pull origin master
cmake -DWITH_CERES=OFF ..
make
```

Sara-Esam, 2021-11-16:
This is a follow-up, 

I was trying to figure it out. 
So first I had this cyclic constraint graph problem which I thought could have anything to do with it. 
Although the compiler ran successfully anyway, I found out that it's because of this:  `sudo mv /usr/local/lib/CMake/* /usr/local/lib/cmake/.`

```
CMake Warning at corelib/src/CMakeLists.txt:729 (ADD_LIBRARY):
  Cannot generate a safe runtime search path for target rtabmap_core because
  there is a cycle in the constraint graph:

    dir 0 is [/home/xliu/rtabmap/bin]
    dir 1 is [/usr/local/lib]
      dir 0 must precede it due to runtime library [librtabmap_utilite.so.0.20]
      dir 4 must precede it due to runtime library [libg2o_core.so]
      dir 6 must precede it due to runtime library [libbrisk.so]
    dir 2 is [/usr/lib/x86_64-linux-gnu/hdf5/openmpi]
    dir 3 is [/usr/lib/x86_64-linux-gnu/openmpi/lib]
    dir 4 is [/opt/ros/melodic/lib]
    dir 5 is [/usr/local/zed/lib]
    dir 6 is [/home/xliu/okvis/build/lib]
      dir 1 must precede it due to runtime library [libceres.so.1]

  Some of these libraries may not be found correctly.

```

Now the issue is obviously due to the incompatibility of ceres and eigen3. 

I found here ( https://githubmemory.com/repo/ceres-solver/ceres-solver/issues/690) that  Ceres 1.9.0 requires older versions of Ubuntu and Eigen to run. Unless necessary, it is recommended to use modern Ceres versions. 

Have you ever used OKVIS on Ubuntu 18.04 with Eigen 3.3.4?  If so, could you tell me what settings can fix this issue? 

Otherwise, I will close the issue.

Sara-Esam, 2021-11-16:
This solves the problem. 
Thank you so much for your help!

closed 2021-11-16 by Sara-Esam

Serafadam, 2024-03-15:
Hi @matlabbe, I tried building okvis same way as mentioned above but it seems that those Cmake lines were changed back, is there a way to return to that old behavior?

matlabbe, 2024-03-17:
I reverted the change made for floam in https://github.com/introlab/rtabmap/commit/264c560a2c43d7abbc0563a05d96adf77841ba9c

Serafadam, 2024-03-18:
Thanks! Just one quick question, when is the next release scheduled?

matlabbe, 2024-03-24:
It can be sooner if someone needs it. However, there are currently not that many differences from the latest release for the standalone version right now. I may wait for https://github.com/introlab/rtabmap/pull/1193 to justify a new standalone release. In the mean time, I created that tag [0.21.4.1](https://github.com/introlab/rtabmap/releases/tag/0.21.4.1) [yesterday](https://github.com/introlab/rtabmap/issues/1250) for vcpkg folks, if you prefer a tag instead a commit version.