introlab/rtabmap #785: Can't build library on macOS

romsahel, 2021-11-15:
Hello,

I'm trying to build the iOS app from source but I'm stuck at the installation step of the library. 
I'm following the [wiki](https://github.com/introlab/rtabmap/wiki/Installation#macosx) and installed everything mentionned without issues (even the optional packages, just in case) but building of the actual `rtabmap` library leads to complications.
For the `cmake` command I had to specify a few path (namely, `Qt5_DIR` and `OpenCV_DIR`) but I'm stuck at the following step where I get the following error:
```
/usr/local/include/QtCore/qglobal.h:96:6: error: "Qt requires a C++17 compiler"
``` 

I've tried to explicitly ask `cmake` for a C++17 compilation (`-DCMAKE_CXX_FLAGS="-std=c++17"` or `-DCXX_STANDARD=17`) or to exclude Qt from the compilation (`-DWITH_QT=OFF`) but to no avail. Am I using the wrong QT version? I believe it was installed automatically as a dependency when installing other libraries (as this computer was mostly clean).
Any help would be much appreciated,
Thanks,

matlabbe, 2021-11-15:
For the MacOS build, the homebrew recipes or default versions may have changed. I will need to update my Mac to check your dependency problems.

However, you said iOS app, which should be built differently from MacOS instructions. There are no official instructions to build RTAB-Map iOS, there is an xcode project here: https://github.com/introlab/rtabmap/tree/master/app/ios
and dependencies are built using this script: https://github.com/introlab/rtabmap/blob/master/app/ios/RTABMapApp/Libraries/install_deps.sh (tested only on Mac Apple Silicon)

matlabbe, 2021-11-15:
Found a way to build a working RTAB-Map on MacOS using latest homebrew (to this date). We can build RTAB-Map with these dependencies:
```bash
brew install pcl opencv cmake
# For issue cmake not finding Qt5 correctly (adjust qt version to yours!):
brew link --force qt5
ln -s $HOMEBREW_CELLAR/qt@5/5.15.2_1/mkspecs $HOMEBREW_PREFIX/mkspecs
ln -s $HOMEBREW_CELLAR/qt@5/5.15.2_1/plugins $HOMEBREW_PREFIX/plugins
```
However you will get this error on Mac M1 (maybe on Mac Intel the above would be okay):
`
vtkOpenGLRenderWindow.c:506    ERR| vtkGenericOpenGLRenderWindow (0x12e158510): Unable to find a valid OpenGL 3.2 or later implementation. Please update your video card driver to the latest version. If you are using Mesa please make sure you have version 11.2 or later and make sure your driver in Mesa supports OpenGL 3.2 such as llvmpipe or openswr. If you are on windows and using Microsoft remote desktop note that it only supports OpenGL 3.2 with nvidia quadro cards. You can use other remoting software such as nomachine to avoid this issue.
`

The workaround with minimal recompilation is to rebuild pcl with vtk8. To do this (from a clean homebrew directory):
```bash
brew install qt@5 opencv@3 cmake vtk@8.2 boost cminpack eigen flann glew libomp libpcap libusb qhull
brew link vtk@8.2 opencv@3
# For issue cmake not finding Qt5 correctly (adjust qt version to yours!):
brew link --force qt5
ln -s $HOMEBREW_CELLAR/qt@5/5.15.2_1/mkspecs $HOMEBREW_PREFIX/mkspecs
ln -s $HOMEBREW_CELLAR/qt@5/5.15.2_1/plugins $HOMEBREW_PREFIX/plugins

git clone https://github.com/PointCloudLibrary/pcl
cd pcl
mkdir build
cd build
cmake ..
make
sudo make install
```
We install/uninstall PCL to install all its dependencies, then we remove PCL and VTK9. We need to explicitly set the sym links for vtk@8.2 and opencv@3, and fix Qt5 syn links too. Then build rtabmap as usual.

romsahel, 2021-11-16:
Thanks for you detailed and quick answer.
It does help: I just had to add the `numpy` folder to my `include` folder to prevent an error from interrupting the build (`fatal error: 'numpy/arrayobject.h' file not found`).
Now, it does compile and `make install` executes successfully 👍 .
As a side note, I did not need the workaround you mention above (I do have a Mac Intel).

However, as you understood, I am actually trying to run the iOS app rather than the MacOS (sorry for the confusion), I just presumed it first needed the MacOS `Rtabmap` correctly installed (I had tried to run the `install_deps.sh` script but it failed as well).
Now looking into it more:
1. I installed `openmp` from `brew`
2. I replaced the `wget` to the `flann` zip by `git clone https://github.com/kjinxin/flann-1.8.4-src.git` (since website seems down, hopefully, it is identical) and I followed the instructions inside the script (commenting test, doc and even examples),
3. I followed other similar instructions specified for other libs
After all that, all dependencies seem to compile fine. However, the first `cmake -G` command for rtabmap fails with:
`rtabmap/bin/rtabmap-res_tool: No such file or directory`. I'm not sure where this is supposed to come from, I don't have a bin/ folder.

matlabbe, 2021-11-16:
Indeed, for iOS build you don't need homebrew or make the MacOS build. I checked again in the install script from a clean repo. I fixed most of your issues (fixed flann broken link and missing `-lhdf5` error, fixed `Undefined symbol: _png_init_filter_functions_neon` error) in the commit above. I made a new section in installation instructions on this page: https://github.com/introlab/rtabmap/wiki/Installation#ios

As a side note for MacOS, the numpy error is related to python version. I commited [here](https://github.com/introlab/rtabmap/commit/9ae2c465465d1fcb8824f3a14d40a74ac75d84f3) a fix to really ignore python if not explicitly set by WITH_PYTHON option, until I find why the numpy include path was not added in recent python versions.

romsahel, 2021-11-17:
Excellent! I pulled your fixes and re-ran the `install_deps.sh` and it worked (though I did not clean the previous build cache).
Thank you very much for you quick and efficient help!

closed 2021-11-17 by romsahel

Ria8651, 2022-01-19:
For anyone trying the M1 workaround, PCL won't compile with the latest commit in the PCL repository. After cloning the repository run `git checkout af3a2ea43447c9e0d2b3757fedb8f5a09d3ea93e`. You should probably always add a specific commit when linking to a repository as things are constantly changing. It'd be good to have this added to the wiki too.

matlabbe, 2022-01-20:
Just tried with PCL 1.12.1, is this the same error you had?
```
[ 26%] Linking CXX shared library ../lib/libpcl_io.dylib
ld: library not found for -lvtkIOMPIImage
```
Looks like indeed an incompatibility with VTK8 used in the instructions. PCL+VTK9 will eventually work on M1, though I cannot test it everytime they push a new commit. As you suggested, I [updated the wiki](https://github.com/introlab/rtabmap/wiki/Installation#source) with the actual git versions I used for all packages built from source. Thx!

Ria8651, 2022-01-20:
Indeed that was the error I had. I also had an error when building RTABMap so I just used an old commit - I can't remember what the error was. But it turns out Orbbec dropped support for mac and I can't find the old OpenNI drivers that used to work so I can't actually use RTABMap 😕

rlabs-oss, 2022-02-23:
Tried to build and get the vtk opengl issue, 
[          2FADC6]vtkOpenGLRenderWindow.c:506    ERR| vtkGenericOpenGLRenderWindow (0x1301a7010): Unable to find a valid OpenGL 3.2 or later implementation. Please update your video card driver to the latest version. If you are using Mesa please make sure you have version 11.2 or later and make sure your driver in Mesa supports OpenGL 3.2 such as llvmpipe or openswr. If you are on windows and using Microsoft remote desktop note that it only supports OpenGL 3.2 with nvidia quadro cards. You can use other remoting software such as nomachine to avoid this issue.

building pcl results in:-
[ 97%] Building CXX object tools/CMakeFiles/pcl_mesh2pcd.dir/mesh2pcd.cpp.o
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/lib/libz.tbd', needed by `bin/pcl_mesh2pcd'.  Stop.
make[1]: *** [tools/CMakeFiles/pcl_mesh2pcd.dir/all] Error 2
make: *** [all] Error 2

Based on the last build cleaning out brew and installing in a specific order fixed things but Im unable to get past the pcl building problem.

matlabbe, 2022-02-24:
It seems failing while building PCL tools. Try to disable pcl tools with `cmake BUILD_tools=OFF ..`.

rlabs-oss, 2022-02-24:
Seems to be an invalid configuration of the SDK which may have changed since pcl? - not sure its a good idea to add a symlink from MacOS12.sdk to MacOSX.sdk to resolve this since it shouldnt be using that in the first place based on the correct cmake sysroot?

Consolidate compiler generated dependencies of target pcl_mesh2pcd
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/lib/libz.tbd', needed by `bin/pcl_mesh2pcd'.  Stop.
make[1]: *** [tools/CMakeFiles/pcl_mesh2pcd.dir/all] Error 2
make: *** [all] Error 2

The lib its looking for is elsewhere-

build % ls -la /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib/libz.tbd 
lrwxr-xr-x  1 root  wheel  10 Apr 30  2021 /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib/libz.tbd -> libz.1.tbd

cmake sysroot is using another location
sysroot=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk

Chieken-Nug, 2026-03-06:
Not sure if this is useful guys, but for opengl issues i did a custom compile with the lib freenect flag. Not sure if this is allowed, but i have a public repo on my profile, rtab_map_freenect, compiled on macOS Sonoma on M1. Not sure if it will work on anyone else machines but if you want to i think it's worth a try.