introlab/rtabmap #786: Crashes in iOS 15

Zumbalamambo, 2021-11-22:
The application crashes in a few seconds after I triggered scanning. Also, when it loses track, it doesn't resume. Are there any possible fixes? I tried to uninstall and reinstall but still, I have got the same issues. 

Is it possible to use the input from 3d scanner app to create the clean 3d model?

matlabbe, 2021-11-23:
I cannot reproduce the crash at the beginning on iOS 15.1 (tested on ipad 2020). Is it the version from the AppleStore? What is your iPhone/iPad model? Did you try rebooting your device?

If it ARKit loses tracking or drift a lot, it is recommended to stop the scan, then click on "Append". Press record and point to a previously scanned area with same point of view, the scanning should then continue from there (see also at 4:13 in this [video](https://youtu.be/rVpIcrgD5c0)):
[image]

For 3D Scanner App exported folder, normally [this](https://github.com/introlab/rtabmap/issues/632#issuecomment-748525481) would work. However, I checked today and 3D Scanner App doesn't save JPG image for all frames, so you would have to manually remove the json and depth files not having corresponding rgb frame. I updated this [post](https://github.com/introlab/rtabmap/issues/632#issuecomment-976076592) to show this.

If you want to use the mesh created from 3D SCanner App inside RTAB-Map, it is not possible.

matlabbe, 2021-11-25:
How did you get this log? Is it the version from the AppleStore? What is your iPhone/iPad model? Did you try rebooting your device (maybe the RAM is full)?

Zumbalamambo, 2021-11-25:
@matlabbe I got this error when I tried to process the data from 3D Scanner app from the desktop version of RTABMap

matlabbe, 2021-11-25:
If you are using the standalone version of RTAB-Map, set in Preferences->Logging-> level to Debug. It will be easier to see from which code inside rtabmap the call to opencv failed. Also which desktop version are you using? on which OS?

Zumbalamambo, 2021-11-25:
It throws the following error when I enabled logging,

```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.4-dev) /home/opencv/opencv/modules/core/src/matrix.cpp:250: error: (-215:Assertion failed) s >= 0 in function 'setSize'

Aborted (core dumped)
```

I'm using Ubuntu 20.0.4 and I use 0.20.13

matlabbe, 2021-11-25:
Humm, this is not system OpenCV version that is loaded. Is rtabmap built from source with that opencv version? Otherwise if you are using rtabmap binaries, then have this OpenCV built from source found first by the linker on runtime, it is likely that rtabmap will crash because it should be using OpenCV 4.2.0 version instead on the computer.

Can you show the cmake output if you are building rtabmap from source? Also show "ldd rtabmap".

Zumbalamambo, 2021-11-29:
this is the output of `ldd rtabmap`.
```
	linux-vdso.so.1 (0x00007ffce913a000)
	librtabmap_gui.so.0.20 => /usr/local/lib/librtabmap_gui.so.0.20 (0x00007f2448bed000)
	librtabmap_core.so.0.20 => /usr/local/lib/librtabmap_core.so.0.20 (0x00007f2448446000)
	librtabmap_utilite.so.0.20 => /usr/local/lib/librtabmap_utilite.so.0.20 (0x00007f2448423000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f2447d88000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f244783f000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f244765e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2447641000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f244744f000)
	libpcl_surface.so.1.10 => /lib/x86_64-linux-gnu/libpcl_surface.so.1.10 (0x00007f24466c1000)
	libpcl_visualization.so.1.10 => /lib/x86_64-linux-gnu/libpcl_visualization.so.1.10 (0x00007f244654e000)
	libvtkRenderingLOD-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingLOD-7.1.so.7.1p (0x00007f244652d000)
	libvtkGUISupportQt-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkGUISupportQt-7.1.so.7.1p (0x00007f24464d7000)
	libQt5Svg.so.5 => /lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007f2446479000)
	libQt5PrintSupport.so.5 => /lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007f2446400000)
	libopencv_imgcodecs.so.4.5 => /usr/local/lib/libopencv_imgcodecs.so.4.5 (0x00007f244639f000)
	libopencv_calib3d.so.4.5 => /usr/local/lib/libopencv_calib3d.so.4.5 (0x00007f244616c000)
	libopencv_imgproc.so.4.5 => /usr/local/lib/libopencv_imgproc.so.4.5 (0x00007f24445f9000)
	libopencv_core.so.4.5 => /usr/local/lib/libopencv_core.so.4.5 (0x00007f2442b52000)
	libpcl_io.so.1.10 => /lib/x86_64-linux-gnu/libpcl_io.so.1.10 (0x00007f24428c9000)
	libpcl_search.so.1.10 => /lib/x86_64-linux-gnu/libpcl_search.so.1.10 (0x00007f2442698000)
	libpcl_common.so.1.10 => /lib/x86_64-linux-gnu/libpcl_common.so.1.10 (0x00007f24425e4000)
	libvtkRenderingAnnotation-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingAnnotation-7.1.so.7.1p (0x00007f2442411000)
	libvtkRenderingFreeType-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingFreeType-7.1.so.7.1p (0x00007f244234f000)
	libvtkInteractionStyle-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkInteractionStyle-7.1.so.7.1p (0x00007f24422b5000)
	libvtkRenderingOpenGL2-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingOpenGL2-7.1.so.7.1p (0x00007f2441f8c000)
	libvtkRenderingCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingCore-7.1.so.7.1p (0x00007f2441c7b000)
	libvtkFiltersSources-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkFiltersSources-7.1.so.7.1p (0x00007f2441b9b000)
	libvtkFiltersGeneral-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkFiltersGeneral-7.1.so.7.1p (0x00007f2441851000)
	libvtkCommonExecutionModel-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonExecutionModel-7.1.so.7.1p (0x00007f2441788000)
	libvtkCommonDataModel-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonDataModel-7.1.so.7.1p (0x00007f24413b0000)
	libvtkCommonMath-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonMath-7.1.so.7.1p (0x00007f2441387000)
	libvtkCommonCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonCore-7.1.so.7.1p (0x00007f244103b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2441018000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f2440a2d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f24408de000)
	libopencv_stitching.so.4.5 => /usr/local/lib/libopencv_stitching.so.4.5 (0x00007f2440760000)
	libopencv_aruco.so.4.5 => /usr/local/lib/libopencv_aruco.so.4.5 (0x00007f24406fe000)
	libopencv_cudafeatures2d.so.4.5 => /usr/local/lib/libopencv_cudafeatures2d.so.4.5 (0x00007f243e65d000)
	libopencv_xfeatures2d.so.4.5 => /usr/local/lib/libopencv_xfeatures2d.so.4.5 (0x00007f243e255000)
	libpcl_segmentation.so.1.10 => /lib/x86_64-linux-gnu/libpcl_segmentation.so.1.10 (0x00007f243d4db000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f243d4bf000)
	libsqlite3.so.0 => /lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f243d381000)
	libOpenNI2.so.0 => /lib/libOpenNI2.so.0 (0x00007f243d31b000)
	librealsense2.so.2.45 => /lib/x86_64-linux-gnu/librealsense2.so.2.45 (0x00007f243bed1000)
	libdc1394.so.22 => /lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007f243be59000)
	libg2o_csparse_extension.so => /usr/local/lib/libg2o_csparse_extension.so (0x00007f243be53000)
	libg2o_types_slam2d.so => /usr/local/lib/libg2o_types_slam2d.so (0x00007f243bde6000)
	libg2o_types_sba.so => /usr/local/lib/libg2o_types_sba.so (0x00007f243bd6c000)
	libgtsam.so.4 => /lib/x86_64-linux-gnu/libgtsam.so.4 (0x00007f243b831000)
	libopencv_videoio.so.4.5 => /usr/local/lib/libopencv_videoio.so.4.5 (0x00007f243b797000)
	libopencv_video.so.4.5 => /usr/local/lib/libopencv_video.so.4.5 (0x00007f243b70d000)
	libopencv_features2d.so.4.5 => /usr/local/lib/libopencv_features2d.so.4.5 (0x00007f243b630000)
	libopencv_photo.so.4.5 => /usr/local/lib/libopencv_photo.so.4.5 (0x00007f243b2cd000)
	libpcl_features.so.1.10 => /lib/x86_64-linux-gnu/libpcl_features.so.1.10 (0x00007f2438b51000)
	libpcl_filters.so.1.10 => /lib/x86_64-linux-gnu/libpcl_filters.so.1.10 (0x00007f2438335000)
	libpcl_sample_consensus.so.1.10 => /lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.10 (0x00007f2437643000)
	libpcl_kdtree.so.1.10 => /lib/x86_64-linux-gnu/libpcl_kdtree.so.1.10 (0x00007f24374ea000)
	libgomp.so.1 => /lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f24374a5000)
[... cut ...]
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f23d8882000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f23d887b000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f23d886c000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f23d8813000)
	libpcre2-8.so.0 => /lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007f23d8783000)
	libdatrie.so.1 => /lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f23d8779000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f23d8772000)

```

Zumbalamambo, 2021-12-20:
i give up.

matlabbe, 2021-12-20:
What I would do is to uninstall opencv built from source, then build rtabmap with OpenCV binaries instead. If it works, then the issue is a compatibility issue with latest opencv version.

closed 2021-12-22 by Zumbalamambo