introlab/rtabmap #789: iOS Build failed because ULogger  not found

WiiLin, 2021-11-26:
[image]

matlabbe, 2021-11-29:
Did you follow this? https://github.com/introlab/rtabmap/wiki/Installation#ios
or the `install.sh` has failed?

diandrakathyusha, 2022-10-19:
Hi, I have the same exact issue. I have followed the instruction. And yes, I think my install.sh has failed. I am using MacOS Monterey on a Mac Mini 2014. Do you have any other alternative than the `$ ./install.sh` ?

matlabbe, 2022-11-26:
Ohhh, I didn't try to build on any Intel Mac. This may be the issue as I just tried again on my Mac Mini M1 and the script worked.

matlabbe, 2023-03-25:
I think it is a duplicate of https://github.com/introlab/rtabmap/issues/998, will close as it is fixed.

closed 2023-03-25 by matlabbe