introlab/rtabmap #792: cannot build with last rtabmap version 20.16

FPSychotic, 2021-11-28:
i could with 20.14 and 20.15 without issues

`[ 60%] Generating ui_exportCloudsDialog.h
[ 61%] Generating ui_exportDialog.h
[ 61%] Generating ui_loopClosureViewer.h
[ 61%] Generating ui_mainWindow.h
[ 61%] Generating ui_postProcessingDialog.h
Scanning dependencies of target rtabmap_gui
[ 62%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/MainWindow.cpp.o
[ 62%] Linking CXX executable ../../bin/rtabmap-camera
/usr/bin/ld: CMakeFiles/camera.dir/main.cpp.o: in function `main':
main.cpp:(.text.startup+0x200): undefined reference to `ULogger::kDefaultLogFileName'
/usr/bin/ld: main.cpp:(.text.startup+0x220): undefined reference to `ULogger::kDefaultLogFileName'
/usr/bin/ld: main.cpp:(.text.startup+0x24c): undefined reference to `ULogger::setType(ULogger::Type, std::string const&, bool)'
/usr/bin/ld: main.cpp:(.text.startup+0x2cc): undefined reference to `uStr2Float(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x3ac): undefined reference to `UFile::exists(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x3c4): undefined reference to `UFile::getExtension(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x4e0): undefined reference to `UDirectory::getDir(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x4f0): undefined reference to `UFile::getName(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x61c): undefined reference to `UDirectory::exists(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x798): undefined reference to `cv::namedWindow(std::string const&, int)'
/usr/bin/ld: main.cpp:(.text.startup+0x800): undefined reference to `cv::destroyWindow(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x860): undefined reference to `cv::imshow(std::string const&, cv::_InputArray const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uNumber2Str(float)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointXYZI>::Search(std::string const&, bool)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `g2o::Factory::registerType(std::string const&, g2o::AbstractHyperGraphElementCreator*)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uToUpperCase(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::Factor::printKeys(std::string const&, std::function<std::string (unsigned long)> const&) const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointNormal>::Search(std::string const&, bool)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `g2o::Factory::unregisterType(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `UDirectory::makeDir(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::PCDWriter::resetLockingPermissions(std::string const&, boost::interprocess::file_lock&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::imread(std::string const&, int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `UDirectory::getNextFileName()'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::FileStorage::FileStorage(std::string const&, int, std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `UFile::length(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::Unit3::print(std::string const&) const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `UDirectory::separator()'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::FileStorage::writeRaw(std::string const&, void const*, unsigned long)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::FileNode::string() const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uNumber2Str(double)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointXYZRGBNormal>::getName() const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::error(int, std::string const&, char const*, char const*, int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::_multirobotKeyFormatter(unsigned long)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::ONIGrabber::ONIGrabber(std::string const&, bool, bool)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::FileNodeIterator::readRaw(std::string const&, void*, unsigned long)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointXYZRGBNormal>::Search(std::string const&, bool)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uNumber2Str(int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::PCDWriter::writeBinary(std::string const&, pcl::PCLPointCloud2 const&, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `dai::Device::getOutputQueue(std::string const&, unsigned int, bool)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::NoiseModelFactor::print(std::string const&, std::function<std::string (unsigned long)> const&) const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::write(cv::FileStorage&, std::string const&, int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `UVariant::toStr(bool*) const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::imwrite(std::string const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::PCDReader::read(std::string const&, pcl::PCLPointCloud2&, Eigen::Matrix<float, 4, 1, 0, 4, 1>&, Eigen::Quaternion<float, 0>&, int&, int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `octomap::AbstractOccupancyOcTree::writeBinary(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::Rot2::print(std::string const&) const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uReplaceChar(std::string const&, char, std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::_defaultKeyFormatter(unsigned long)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uReplaceChar(std::string const&, char, char)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `UDirectory::UDirectory(std::string const&, std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uHex2Str(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::io::saveVTKFile(std::string const&, pcl::PCLPointCloud2 const&, unsigned int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uNumber2Str(unsigned int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointXYZI>::getName() const'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `gtsam::demangle(char const*)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::VideoCapture::open(std::string const&, int)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `pcl::search::Search<pcl::PointXYZRGB>::Search(std::string const&, bool)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::getBuildInformation()'
[... 31 lines cut ...]
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `uStr2Double(std::string const&)'
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.16: undefined reference to `cv::imencode(std::string const&, cv::_InputArray const&, std::vector<unsigned char, std::allocator<unsigned char> >&, std::vector<int, std::allocator<int> > const&)'
[... cut ...]
/usr/bin/ld: main.cpp:(.text.startup+0x7a4): undefined reference to `uReplaceChar(std::string const&, char, std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x800): undefined reference to `uReplaceChar(std::string const&, char, std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0xa98): undefined reference to `uStr2Double(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0xb00): undefined reference to `uStr2Double(std::string const&)'
/usr/bin/ld: main.cpp:(.text.startup+0x14e0): undefined reference to `cv::namedW`

FPSychotic, 2021-11-30:
I could compile 20.14 again, but now rtabmap_ros doesnt install with it (something related with gtam,maybe a new udate from ppa?). I removed any file, resourced etc. nothing works, everything started having 0.20.14 working in ros1 noetic 20.04 and 0.20.15 in foxy, git pull, and everything went broken.

FPSychotic, 2021-12-01:
the problem was that with open3d , wont compile with it

matlabbe, 2021-12-05:
`WITH_OPEN3D` is OFF by default. Do I understand with your last comment that it is fine with `-DWITH_OPEN3D=OFF` but not `-DWITH_OPEN3D=ON` ?

FPSychotic, 2021-12-06:
Yes it works with open3d off. I have it compiled in a ARM64 ketdom Xavier so it is experimental.
Maybe is that

matlabbe, 2021-12-07:
Open3D support inside rtabmap is quite experimental too, it is why it is disabled by default. I'll close this issue for the original error.

closed 2021-12-07 by matlabbe