introlab/rtabmap #795: DISABLE_VTK not defined in new version of PCL library

windelbouwman, 2021-12-10:
When compiling rtabmap with a newer version of the PCL library, the define DISABLE_VTK is not defined. This causes a compilation error in rtabmap. 

For example, this piece of code does not compile when using the PCL library 1.10: https://github.com/introlab/rtabmap/blob/master/corelib/src/Parameters.cpp#L44

What's notable here, is that we use `ifndef DISABLE_VTK`, which is a double negation. Might it be good to use something like `ifdef WITH_VTK`, if this exists?

matlabbe, 2022-05-28:
The problem is how can we detect if PCL has been built with VTK or not? DISABLE_VTK is used not to check only if VTK is there, but also if PCL's related functions using VTK are there, like here:
https://github.com/introlab/rtabmap/blob/656da152b31e4596235600b111d4d48530066e07/corelib/src/util3d_surface.cpp#L61-L63

Maybe there is a new define to let us know if PCL is built with VTK or not.

matlabbe, 2022-05-28:
Looking this: https://github.com/PointCloudLibrary/pcl/blob/c509b2ba0d7866b9d8353387cf9e8126eab03d93/surface/CMakeLists.txt#L31-L48

We may assume that if VTK_FOUND is defined, PCL is built with it. I'll check the related pull requests.