introlab/rtabmap #798: segmentation fault when running rtabmap?

jcyhcs, 2021-12-24:
hi,professor:
i compile rtabmap standalone success on nvidia jetson tx2, but when i try to run rtabmap, i got segmentation fault: 
[image]
please HELP ME!

Shishir-Kumar-Singh, 2021-12-25:
> hi,professor: i compile rtabmap standalone success on nvidia jetson tx2, but when i try to run rtabmap, i got segmentation fault: [image] please HELP ME!

@jcyhcs hi, the segmentation fault comes when the linked libraries have version mismatch. So at the time of compiling try to check the path of the VTK, OpenGL and Qt. I assume, you have manually compiled VTK with OpenGL2 and link library is the correct one.

jcyhcs, 2021-12-27:
@Shishir-Kumar-Singh 
thanks professor! i remove all files about qt5,then i re-install qt5 and re-compile rtabmap, then it works fine. i  also flow tutorials, then i save an obj file, like this:
[image]
emm.., i open it with meshlab, but how can i look inside it? for example cut it half , so i can look the inside, like the example 
[image]
i don't know how ? please help!

closed 2021-12-28 by jcyhcs