introlab/rtabmap #799: randomly occur error: terminate called after throwing an instance of 'std::out_of_range'  what():  map::at

cdb0y511, 2021-12-27:
Hi, 
I notice this error when I use T265 + L515 model.
It traces back to 

__GI_raise 0x00007fbcf27fb18b
__GI_abort 0x00007fbcf27da859
<unknown> 0x00007fbcf2a60911
<unknown> 0x00007fbcf2a6c38c
std::terminate() 0x00007fbcf2a6c3f7
__cxa_rethrow 0x00007fbcf2a6c6fd
gtsam::internal::linearAlgorithms::optimizeBayesTree<…>(const gtsam::GaussianBayesTree &) 0x00007fbcebf58d1d
gtsam::GaussianBayesTree::optimize() const 0x00007fbcebf45932
gtsam::GaussianFactorGraph::optimize(const gtsam::Ordering &, const std::function<…> &) const 0x00007fbcebf71042
gtsam::NonlinearOptimizer::solve(const gtsam::GaussianFactorGraph &, const gtsam::NonlinearOptimizerParams &) const 0x00007fbcec04bf0b
gtsam::GaussNewtonOptimizer::iterate() 0x00007fbcec00139b
rtabmap::OptimizerGTSAM::optimize OptimizerGTSAM.cpp:572
rtabmap::Rtabmap::optimizeGraph Rtabmap.cpp:4554
rtabmap::Rtabmap::optimizeCurrentMap Rtabmap.cpp:4469
rtabmap::Rtabmap::process Signature.h:70
rtabmap::RtabmapThread::process OdometryEvent.h:69
rtabmap::RtabmapThread::mainLoop RtabmapThread.cpp:204
UThread::ThreadMain UThread.cpp:283
UThreadC::ThreadMainHandler UThreadC.h:392
start_thread 0x00007fbcefd30609
clone 0x00007fbcf28d7293

rtabmap::OptimizerGTSAM::optimize OptimizerGTSAM.cpp:572:
			
                    try
			{
				optimizer->iterate();
				++it;
			}

If the detect RATE  gets larger (3HZ- 10HZ), it occurs more often.
I am not sure it is the GTSAM bugs or rtabmap itself.
Have any advice to solve it?

matlabbe, 2021-12-27:
Which GTSAM version is used?

cdb0y511, 2021-12-28:
I think it is GTSAM 4.1 Release. I notice there is a recent release.  I update the GTSAM Release 4.1.1. Still, it needs some time to test if this issue is gone. 
Do you think it comes from GTSAM?

matlabbe, 2022-04-27:
Related to https://github.com/introlab/rtabmap/issues/823 ?

cdb0y511, 2022-04-28:
I think so.