introlab/rtabmap #802: [New Feature] about relocation (or loop closure detected).

lamer-afk, 2021-12-30:
I would like to continue developing a new feature based on your code that will be supported：
（The reason is that fisheye camera become very small after undistortion）

-  Odometry and relocation (or loop closure detected)  do not undistort the full image, only undistort the key points （stereo mode）
-  Create occupancy grid only use laser scan , don't use any cameras 

I think the key modification point should be Memory.createSignature(const SensorData & inputData, const Transform & pose, Statistics * stats)

I think it works in principle. Is that right?

matlabbe, 2022-01-07:
Hi,

For the second point, you have to set Grid/FromDepth=false and feed a laser scan to rtabmap node.

For the first point, it is indeed in createSignature that would have some modifs. I can already accept not rectified images and keypoints, however for stereo, the images would be rectified there https://github.com/introlab/rtabmap/blob/ab50ba0b6cbff667eebca333d909638e6a4178bd/corelib/src/Memory.cpp#L4415 to find stereo correspondences here:
https://github.com/introlab/rtabmap/blob/ab50ba0b6cbff667eebca333d909638e6a4178bd/corelib/src/Memory.cpp#L4804
I think that if you implement a function to find stereo correspondences and find disparity for each distorted keypoints, I could check to integrate it instead of the current stereo rectification approach. So, inside generateKeypints3D above, the stereo approach is defined by those functions (assuming stereo rectified images):
https://github.com/introlab/rtabmap/blob/7e40196439cc5359640e5ebc5b4e72fa519e70ee/corelib/src/Features2d.cpp#L813-L826
The approach looks for each keypoint in left image, it searches for the same point on same image line of right image (using [OpenCV Optical Flow](https://github.com/introlab/rtabmap/blob/c4d127cae40e8f92db07c05c0dadcaef4cb97d14/corelib/src/util2d.cpp#L360-L371)), thus finding disparity required to project the point in 3D. If you can change those functions to do it with raw fisheye images, that could be indeed nice.

cheers,
Mathieu

lamer-afk, 2022-01-07:
Thank you very much for solving my puzzled problem.

closed 2022-01-07 by lamer-afk