introlab/rtabmap #807: iOS View in rviz

controller1304, 2022-01-04:
I am very impressed by the RTAB Map App for iOS. I opened the *.DB File in rviz in a Desktop Ununtu PC. I can see the Pointcloud there only. How can I get the mesh View with applied textures like i see ist in the IOs?
My use case is, that i want to view the photorealistic 3d map on the Ubuntu PC and do distance measurements there.

matlabbe, 2022-01-04:
Do you mean rtabmapviz? You can see the optimized mesh with rtabmap-databaseViewer: File- > View optimized Mesh. You can also export OBJ with File -> Export optimized mesh (OBJ can also be exported directly from iOS app). 

For rviz, we would have to publish the mesh, and use a plugin from [mesh_tools](http://wiki.ros.org/mesh_tools) to show it in rviz. I can take a look at how to do so with that package.

Note that if you need to take some measurements (without ros), you may download MapIt Windows (or Mac OS X) version here: https://introlab.3it.usherbrooke.ca/mediawiki-introlab/index.php/MapIt#Windows
The web page is written in french (it is a local project here for occupational therapists), but the application is both in English and French. It should be able to open databases created from RTAB-Map iOS app (you have to drop your database in "My Documents/MapIt" directory to show up automatically in MapIt).

controller1304, 2022-01-04:
On the rtabmap-databaseViewer I can see the pointcloud only. After opening the *db file i cannot choose View optimized Mesh. On the iOS app I cannot export the optimized mesh.

It would be perfect for me if you could give me an idea how to generate a mesh on rviz like in the iOS app with the mesh_tools plugin for rviz.

I installed MapIt on a Windows computer, but was not able to open the *db file saved in the iOS app. It says: "Failed to open 1 darabases (too old MapIt version?). They won't appear in the list view.

matlabbe, 2022-01-04:
> After opening the *db file i cannot choose View optimized Mesh. On the iOS app I cannot export the optimized mesh.

This means the database doesn't contain an optimized mesh. Make sure on iOS to do Assemble...->Optimized Mesh...->Textured Mesh-> default 200 000 polygons -> ok. Then do "Save". Note also during visualization of the optimized mesh, there is an Export OBJ/PLY button that should appear if you want to export form the App. Otherwise, the database should now contain the optimized Mesh that can be viewed in rtabmap-databaseViewer as explained above. For MapIt error, it is maybe related to optimized mesh not saved in the database, which would work after generating it as I just explained. IF still not working, there is maybe a version compatibility issue between those apps.

For mesh_tools, it is currently not supported from rtabmap node, that feature should be added.

Regards,
Mathieu

controller1304, 2022-01-05:
Now I got my usecase workin. 

The iOS app always hanged, when I optimized the mesh as desicribed.
But I was able to open the exported *.db file in rtabmap-databaseViewer, optimized the mesh there and saved it (*.obj and *.png). Then opened it in MapIt and there I see the 3D Model in the same quality as in the iOS and can do measurements. The accurancy is amazing it is around 2 centimiters exact.

closed 2022-01-05 by controller1304