introlab/rtabmap #809: How do I do let robot start from at map(0,0,0) postion on navigation mode(localization:=true) start?

tosa-no-onchan, 2022-01-06:
How do I do let robot start from at map(0,0,0) postion on navigation mode(localization:=true) start? 

I want to let  robot start from at tf-map(0,0,0) or tf-odom(0,0,0) when navigation mode start.
But, Turtlebot3 actual machine robot is placed at or starts from at the last position of mapping mode saved.  
In Gazeo simulation,Turtlebot3 robot is placed at the map(0,0,0) when ratbamap_ros is navigation mode starting.
How do I do let acutual robot start from at the tf-map(0,0,0) or the tf-odom(0,0,0) postion ?

matlabbe, 2022-01-07:
Try `RGBD/StartAtOrigin` set to true. It will restart at the origin of the map, so if the first node of the map is at (0,0,0), then the robot will start at (0,0,0).

tosa-no-onchan, 2022-01-07:
Thank You  matlabbe.
It worked.
Fine!

closed 2022-01-07 by matlabbe