introlab/rtabmap #814: Crashed frequently on iPhone

EdXric, 2022-01-19:
Hi, very thankful for sharing your excellent work.
I've tried this app on my iPhone 13 pro max with iOS 15.2. However, sometimes I scanned for like 4 or 5 minutes the app suddenly crashed. Sometimes I could scan for 10 more minutes without crashing (I have closed all the other apps). Also, at the optimization stage or the assemble stage the app crashed a lot, even if I saved first and reload the data (also tried reboot).
Another tiny issue is that when there is a system notification pop up the scan will stop.
Could you tell me which device and system are tested for this app?
With regards.

matlabbe, 2022-01-19:
Thank you for your feedback, I am testing mainly on iPad Pro 2020 (A12z chip). Yesterday, I submitted a new 0.20.18 version particularly to fix performance issues on iPhone 13 Pro and iPad Pro M1. That new version would be approved in the next days. 

For the crashes, can you tell me what you are scanning, which size? Here is one issue that could happen on post-processing (but I have difficulty to reproduce it): https://github.com/introlab/rtabmap/issues/803

Hopefully since 0.20.17 version, a recovery feature has been added, to at least being able to recover a session when it crashes. If you don't have it, I recommend to update, or wait for 0.20.18 to be available. I will report here when 0.20.18 is on AppStore.

For the system notification issue, not sure if it can be fixed, as when the App goes to background, we lose ArKit tracking, so we have to stop mapping. Maybe set in AirPlane mode when scanning to make sure no notifications happen. Does it happen often to be very annoying? You can still do "Append Scan" to continue scanning (see at 4:15 on this video: https://www.youtube.com/watch?v=rVpIcrgD5c0).

cheers,
Mathieu

EdXric, 2022-01-21:
Thank you for replying! Looking forward to trying the update! I'm scanning 4 corridors of a floor (15m long, 2m wide for each). Have you tested it on 15.2? Also, should I avoid scanning the same area? Yesterday I noticed that when I walk around for like 3 or 4 loops, at one position (I saw two former visions due to the drift error) the App trying to localize and suddenly crash.

matlabbe, 2022-02-02:
Here is one example of testing (>550 nodes, 117m travelled distance), we can show debug info by clicking Visibility...->Debug:
[image]

I didn't experienced any interruption during the scan, and the model was correctly optimized and assembled using default dialogs after the scan. This was done on iPad Pro 11' 2020 (A12z chip, iOS 15.2.1). Note sure if the bugs you are seeing are related to iPhone. The last update included a big performance boost on latest iPhones and iPads.

For crashes when exporting/assembling the data, if you have a database that is always crashing during the optimizing or assembling, if you can share that database with me with a dropbox/icloud/gdrive link (here if the database doesn't contain private stuff, or at my username (at) gmail), it could help me to fix it.

Mathieu

anhnt2309, 2022-10-25:
Hi @matlabbe, i'm able to replicate the app crash on the 0.20.19 version on AppStore. Running the code from the master branch of this repo for ios the crash reported as : 
```
Domain: IDEDebugSessionErrorDomain
Code: 11
Failure Reason: Message from debugger: Terminated due to memory issue
User Info: {
    DVTErrorCreationDateKey = "2022-10-25 10:41:22 +0000";
    IDERunOperationFailingWorker = DBGLLDBLauncher;
}
```

matlabbe, 2022-11-26:
Which iphone/ipad are you using? You can decrease rendering quality parameters to save RAM to map larger environments. The full data is still saved in the database. Then for exporting, if you recorded too much data, you can still use RTAB-Map's Desktop version to export the data. I refer often to this [tutorial](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango#7-export-the-mesh) for exporting on desktop version.

matlabbe, 2023-02-01:
It is possible to reduce rendering quality (Settings->Rendering->Point Cloud Density to a lower value), show only point cloud instead of the default textured mesh (View button)... this would reduce RAM usage and also CPU usage. Note that changing Rendering parameters doesn't affect the raw data saved by rtabmap, when assembling the optimized mesh, it will use the raw data.