introlab/rtabmap #820: Segmentation fault (core dumped)

kalyank-nicn, 2022-01-31:
RTAB-Map gives segmentation fault while launching in terminal. 
The error log is shown below:
`(gdb) r
Starting program: /usr/local/bin/rtabmap 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc3860700 (LWP 3493)]
[New Thread 0x7fffc305f700 (LWP 3494)]
[New Thread 0x7fffc285e700 (LWP 3495)]
[New Thread 0x7fffc205d700 (LWP 3496)]
[New Thread 0x7fffc185c700 (LWP 3497)]
[New Thread 0x7fffc105b700 (LWP 3498)]
[New Thread 0x7fffc085a700 (LWP 3499)]
[New Thread 0x7fffc0059700 (LWP 3500)]
[New Thread 0x7fffbf858700 (LWP 3501)]
[New Thread 0x7fffbf057700 (LWP 3502)]
[New Thread 0x7fffbe856700 (LWP 3503)]
[New Thread 0x7fffbe055700 (LWP 3504)]
[New Thread 0x7fffbd854700 (LWP 3505)]
[New Thread 0x7fffbd053700 (LWP 3506)]
[New Thread 0x7fffbc852700 (LWP 3507)]
[New Thread 0x7fffbc051700 (LWP 3508)]
[New Thread 0x7fffba823700 (LWP 3509)]
qt5ct: using qt5ct plugin
[New Thread 0x7fffb8e4d700 (LWP 3510)]
[New Thread 0x7fffacf36700 (LWP 3511)]
[New Thread 0x7fffa7fff700 (LWP 3512)]
[New Thread 0x7fffa77fe700 (LWP 3513)]
[New Thread 0x7fffa6ffd700 (LWP 3514)]
[New Thread 0x7fffa67fc700 (LWP 3515)]
[New Thread 0x7fffa5ffb700 (LWP 3516)]
[New Thread 0x7fffa57fa700 (LWP 3517)]
[New Thread 0x7fffa4ff9700 (LWP 3518)]
[New Thread 0x7fff8bfff700 (LWP 3519)]
[New Thread 0x7fff8b7fe700 (LWP 3520)]
[New Thread 0x7fff8affd700 (LWP 3521)]
[New Thread 0x7fff8a7fc700 (LWP 3522)]
[New Thread 0x7fff89ffb700 (LWP 3523)]
[New Thread 0x7fff897fa700 (LWP 3524)]
[New Thread 0x7fff88ff9700 (LWP 3525)]
[New Thread 0x7fff67fff700 (LWP 3526)]
[New Thread 0x7fff677fe700 (LWP 3527)]
[New Thread 0x7fff66ffd700 (LWP 3528)]
[New Thread 0x7fff667fc700 (LWP 3529)]
[New Thread 0x7fff65ffb700 (LWP 3530)]
[New Thread 0x7fff657fa700 (LWP 3531)]
[New Thread 0x7fff64ff9700 (LWP 3532)]
qt5ct: custom style sheet is disabled
qt5ct: D-Bus global menu: no
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x7fff43fff700 (LWP 3533)]
[New Thread 0x7fff437fe700 (LWP 3534)]
[New Thread 0x7fff42ffd700 (LWP 3535)]
[New Thread 0x7fff427fc700 (LWP 3536)]
Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:
`

matlabbe, 2022-02-03:
What is your machine/os?

kalyank-nicn, 2022-02-04:
I am using Lenovo Legion running on Linux Mint which has NVIDIA RTX 3060 as dedicated GPU and AMD RADEON as integrated GPU. The segmentation fault occurs while running RTAB-Map  with integrated GPU but with dedicated GPU there is no problem.

matlabbe, 2022-02-04:
We don't do llvm calls directly in rtabmap, the error is coming from dependencies. Rendering libraries used by rtabmap are Qt and VTK if it is related to GPU. Otherwise, I've read related errors on google (e.g., https://github.com/ponylang/ponyc/issues/2280, https://stackoverflow.com/questions/62901577/how-to-make-the-llvm-opt-tool-to-report-which-analysis-passes-successfully-used, https://opensource.apple.com/source/llvmCore/llvmCore-3425.0.34/docs/Passes.html#instcombine) about clang optimization options (platform specific?).

kalyank-nicn, 2022-02-04:
Thanks Mathieu for  creating such a beautiful platform for the  community. I will continue my work on dedicated GPU for now and maybe try building with GCC later. I will close this issue for now.
Cheers

closed 2022-02-04 by kalyank-nicn