introlab/rtabmap #824: Regarding initializing Localization

sumitsarkar1, 2022-02-08:
I have a pre-build map and in Localization mode I want to start displaying the pose of a robot on a remote compute ONLY when RTabMap flashes yellow or green i.e. a local match or a loop closure match .Which class/variable captures this successful initialization of localization (given a map)

matlabbe, 2022-02-08:
You could look at the `loopClosureId` (green) or `proximityDetectionId` (yellow) members of http://docs.ros.org/en/lunar/api/rtabmap_ros/html/msg/Info.html (/`rtabmap/info` topic), which witll be !=0 when it happens.

sumitsarkar1, 2022-02-08:
and for the standalone RTabMap

matlabbe, 2022-02-14:
If you catch RtabmapEvent event, the statistics contain 
https://github.com/introlab/rtabmap/blob/578c19cc38d11f6542ab1dc351cff7a9fa646fec/corelib/include/rtabmap/core/Statistics.h#L266-L269

closed 2022-02-15 by sumitsarkar1