introlab/rtabmap #825: rtabmap with depthai

nickjrz, 2022-02-08:
I am currently trying to build rtabmap with depthai and it seems that it succeeds at finding it, see picture attached. But when I check `rtabmap --version` it shows as Depthai = false and driver for OAK-D is disabled on the camera drivers. I am also specifying path of installation for depthai-core on the cmake command, depthai core has been built with depthai shared too. Any insight on what this behavior could be?

[image]
[image]

matlabbe, 2022-02-08:
This is version 0.20.16 launching, if you used latest master, it should be 0.20.19. You can do `which rtabmap` to know which one you are starting. It is maybe the binaries that are built without depthai. If you want to start the built one, don't forget `./` in `./rtabmap`.

nickjrz, 2022-02-09:
I found the issue, it was trying to launch my ros installation instead of the binary built from source, even if doing ./rtabmap . I ended up removing my ros installation and it worked fine, thanks for the help.

closed 2022-02-09 by nickjrz