introlab/rtabmap #826: Csparse installed but not found

jjd9, 2022-02-09:
Hello, first off, love this repo! Great piece of software.
The problem I am having is that while performing mapping and post-processing, I constantly get this warning:
```
 g2o is not built with csparse, so it cannot be used as solver. Using PCG instead.
```
However, I do have csparse installed (via libsuitesparse-dev) and I did build g2o with csparse (I also confirmed this by checking the make output). I am not sure why rtabmap is not detecting it though. Any thoughts on how I can troubleshoot this?

I am on Ubuntu 20.04. I installed RTABMAP using these instructions: https://github.com/introlab/rtabmap/wiki/Installation#ubuntu

matlabbe, 2022-02-14:
The warning is coming from here:
https://github.com/introlab/rtabmap/blob/03a106ea36c5472435dae4d2f967713e77d1dc4e/corelib/src/optimizer/OptimizerG2O.cpp#L189-L195

The G2O_HAVE_CSPARE is defined in g2o include "config.h", here an example on ubuntu 18.04:
```cpp
/opt/ros/melodic/include/g2o$ cat config.h 
#ifndef G2O_CONFIG_H
#define G2O_CONFIG_H

#define G2O_HAVE_OPENGL 1
#define G2O_OPENGL_FOUND 1
/* #undef G2O_OPENMP */
#define G2O_SHARED_LIBS 1
#define G2O_LGPL_SHARED_LIBS 1

// available sparse matrix libraries
#define G2O_HAVE_CHOLMOD 1
#define G2O_HAVE_CSPARSE 1

/* #undef G2O_NO_IMPLICIT_OWNERSHIP_OF_OBJECTS */

#ifdef G2O_NO_IMPLICIT_OWNERSHIP_OF_OBJECTS
#define G2O_DELETE_IMPLICITLY_OWNED_OBJECTS 0
#else
#define G2O_DELETE_IMPLICITLY_OWNED_OBJECTS 1
#endif

/* #undef G2O_SINGLE_PRECISION_MATH */
#ifdef G2O_SINGLE_PRECISION_MATH
    #define G2O_NUMBER_FORMAT_STR "%g"

    #ifdef __cplusplus
        using number_t = float;
    #else
        typedef float number_t;
    #endif
#else
    #define G2O_NUMBER_FORMAT_STR "%lg"

    #ifdef __cplusplus
        using number_t = double;
    #else
        typedef double number_t;
    #endif
#endif

#define G2O_CXX_COMPILER "GNU /usr/lib/ccache/c++"

#ifdef __cplusplus
#include <g2o/core/eigen_types.h>
#endif

#endif
```

If you build g2o from source, check if the config.h of g2o has that define. If yes, maybe rtabmap is building against another config.h somewhere on the computer that doesn't have that define.

jjd9, 2022-02-17:
Ahh you are correct! I had a separate version of g2o installed from another project. That fixed it! Thank you! :smiley:

closed 2022-02-17 by jjd9