introlab/rtabmap #828: DepthAI camera initialization error

nickjrz, 2022-02-10:
Whenever I try to launch rtabmap-dataRecorder or just rtabmap with my OAK-D, it fails at initializing the camera. These are the errors that I get: 

[image]
[image]

Any idea what these crashes could be?

matlabbe, 2022-02-14:
You may try rebuilding rtabmap with this commit: https://github.com/introlab/rtabmap/commit/c99203bbed1d212fa35ed088665d3742da54cd11 to avoid linking on libopencv_viz.so that includes vtk stuff with not the same version than pcl's vtk dependency, causing crash when the wrong vtk version is used on runtime.

I didn't test if depthai built with opencv will also give indirectly the libopencv_viz.so to rtabmap, which can cause the same problem even if fixed on rtabmap side.

nickjrz, 2022-02-14:
Sounds good, I will give that a try!

nickjrz, 2022-02-22:
UPDATE:

I rebuilt rtabmap with that commit but I am still running into the same error, the rtabmap version built is 2.18.0. it seems to be the same issue with linking libopencv_viz.so

matlabbe, 2022-02-23:
Hi,

First you may try WITH_DEPTHAI=OFF to see if libopencv_viz.so is still linked or not to rtabmap.

Second, if "first" is the problem, you may change this [line](https://github.com/luxonis/depthai-core/blob/c2ccafd5be7c2b0624addaca62f6c52935c9bdc5/cmake/depthaiDependencies.cmake#L73) to something like this (explicitly load minimal components)
https://github.com/introlab/rtabmap/blob/e6a5fe9c26226478e6f94362a4435a39c42bcd45/CMakeLists.txt#L224
maybe:
```cpp
find_package(OpenCV 4 QUIET CONFIG COMPONENTS core imgproc)
```
rebuild/install depth_ai and rebuild rtabmap afterwards.

nickjrz, 2022-02-28:
I tried doing that but it still seems that libopencv_viz.so is still being linked. How were you able to get it working with depthai driver and what versions of pcl and vtk were you using?

matlabbe, 2022-03-07:
My current build is using default versions from Ubuntu 20.04: 
```bash
~/workspace/rtabmap/build/bin$ ldd ./rtabmap 
	linux-vdso.so.1 (0x00007fff60bf8000)
	librtabmap_gui.so.0.20 => /home/mathieu/workspace/rtabmap/build/bin/librtabmap_gui.so.0.20 (0x00007ff707c23000)
	librtabmap_core.so.0.20 => /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.20 (0x00007ff706dbb000)
	libtorch_cpu.so => /home/mathieu/.local/lib/python3.8/site-packages/torch/lib/libtorch_cpu.so (0x00007ff6fc36d000)
	libtorch_cuda.so => /home/mathieu/.local/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so (0x00007ff6ee9e6000)
	libtorch.so => /home/mathieu/.local/lib/python3.8/site-packages/torch/lib/libtorch.so (0x00007ff6ee9df000)
	librtabmap_utilite.so.0.20 => /home/mathieu/workspace/rtabmap/build/bin/librtabmap_utilite.so.0.20 (0x00007ff6ee9bb000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007ff6ee2d3000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007ff6edd8a000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff6edba8000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff6edb8d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff6ed999000)
	libpcl_surface.so.1.10 => /lib/x86_64-linux-gnu/libpcl_surface.so.1.10 (0x00007ff6ecc0b000)
	libpcl_visualization.so.1.10 => /lib/x86_64-linux-gnu/libpcl_visualization.so.1.10 (0x00007ff6eca98000)
	libvtkRenderingLOD-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingLOD-7.1.so.7.1p (0x00007ff6eca77000)
	libvtkGUISupportQt-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkGUISupportQt-7.1.so.7.1p (0x00007ff6eca21000)
	libvtkRenderingVolumeOpenGL2-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL2-7.1.so.7.1p (0x00007ff6ec99a000)
	libQt5Svg.so.5 => /lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007ff6ec93e000)
	libQt5PrintSupport.so.5 => /lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007ff6ec8c5000)
	libopencv_imgcodecs.so.4.2 => /lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.2 (0x00007ff6ec86c000)
	libopencv_calib3d.so.4.2 => /lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2 (0x00007ff6ec699000)
	libpcl_io.so.1.10 => /lib/x86_64-linux-gnu/libpcl_io.so.1.10 (0x00007ff6ec412000)
	libpcl_search.so.1.10 => /lib/x86_64-linux-gnu/libpcl_search.so.1.10 (0x00007ff6ec1df000)
	libpcl_common.so.1.10 => /lib/x86_64-linux-gnu/libpcl_common.so.1.10 (0x00007ff6ec12b000)
	libvtkRenderingAnnotation-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingAnnotation-7.1.so.7.1p (0x00007ff6ebf58000)
	libvtkRenderingFreeType-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingFreeType-7.1.so.7.1p (0x00007ff6ebe96000)
	libvtkInteractionStyle-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkInteractionStyle-7.1.so.7.1p (0x00007ff6ebdfc000)
	libvtkRenderingOpenGL2-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingOpenGL2-7.1.so.7.1p (0x00007ff6ebad5000)
	libvtkRenderingCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkRenderingCore-7.1.so.7.1p (0x00007ff6eb7c2000)
	libvtkFiltersSources-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkFiltersSources-7.1.so.7.1p (0x00007ff6eb6e2000)
	libvtkFiltersGeneral-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkFiltersGeneral-7.1.so.7.1p (0x00007ff6eb398000)
	libvtkCommonExecutionModel-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonExecutionModel-7.1.so.7.1p (0x00007ff6eb2cf000)
	libvtkCommonDataModel-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonDataModel-7.1.so.7.1p (0x00007ff6eaef7000)
	libvtkCommonMath-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonMath-7.1.so.7.1p (0x00007ff6eaed0000)
	libvtkCommonCore-7.1.so.7.1p => /lib/x86_64-linux-gnu/libvtkCommonCore-7.1.so.7.1p (0x00007ff6eab82000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff6eab5f000)
	libopencv_imgproc.so.4.2 => /lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2 (0x00007ff6ea5b8000)
	libopencv_core.so.4.2 => /lib/x86_64-linux-gnu/libopencv_core.so.4.2 (0x00007ff6ea208000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007ff6e9c1d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff6e9ace000)
	libopencv_stitching.so.4.2 => /lib/x86_64-linux-gnu/libopencv_stitching.so.4.2 (0x00007ff6e9a0b000)
	libopencv_aruco.so.4.2 => /lib/x86_64-linux-gnu/libopencv_aruco.so.4.2 (0x00007ff6e999d000)
	libpcl_segmentation.so.1.10 => /lib/x86_64-linux-gnu/libpcl_segmentation.so.1.10 (0x00007ff6e8c23000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ff6e8c07000)
	libsqlite3.so.0 => /lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007ff6e8ade000)
	libc10.so => /home/mathieu/.local/lib/python3.8/site-packages/torch/lib/libc10.so (0x00007ff6e8a6f000)
	libpython3.8.so.1.0 => /lib/x86_64-linux-gnu/libpython3.8.so.1.0 (0x00007ff6e8519000)
	libfreenect.so.0.5 => /lib/x86_64-linux-gnu/libfreenect.so.0.5 (0x00007ff6e8507000)
	libOpenNI2.so => /lib/libOpenNI2.so (0x00007ff6e8281000)
	libfreenect2.so.0.2 => /usr/local/lib/libfreenect2.so.0.2 (0x00007ff6e818b000)
	libk4arecord.so.1.4 => /lib/x86_64-linux-gnu/libk4arecord.so.1.4 (0x00007ff6e7dc2000)
	librealsense2.so.2.50 => /usr/local/lib/librealsense2.so.2.50 (0x00007ff6e556d000)
	libdc1394.so.22 => /lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007ff6e54f5000)
	libdepthai-opencv.so => /home/mathieu/workspace/depthai-core/build/install/lib/libdepthai-opencv.so (0x00007ff6e54d6000)
	libg2o_types_slam2d.so => /opt/ros/noetic/lib/libg2o_types_slam2d.so (0x00007ff6e547d000)
	libg2o_types_sba.so => /opt/ros/noetic/lib/libg2o_types_sba.so (0x00007ff6e5421000)
[... cut ...]
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------

```

nickjrz, 2022-03-07:
So I found the issue, I had depthai-core built as dynamic library and that 's when it would link `libopencv_viz.so` so I tried building the depthai library in static mode and it worked fine.  Thanks for the help!

Also, to feed IMU data from OAK-D, do I need to use one of the VIO appoaches integrated in RTAB-Map (OKVIS, MSCKF-VIO or VINS-Fusion) ?

matlabbe, 2022-03-07:
You can, but rtabmap's F2M odometry can handle IMU too to get the map aligned with gravity (make sure to update rtabmap, I fixed imu orientation yesterday with latest depthai-core version). IMUs are published by default. But if you want a more tightly coupled VIO approach, you would have to build with one of those approaches. Refer to this [docker file](https://github.com/introlab/rtabmap/blob/ff739a98a5b0fdee4532b1a4e70ffb77677d59d1/docker/jfr2018/latest/Dockerfile#L124-L150) to see which version of these approaches have been tested (known to build with rtabmap).

nickjrz, 2022-03-09:
Thanks for your prompt reply. I will test the odometry approaches now that I have rtabmap working with the DepthAI driver. Are the VIO approaches also able to be integrated in the rtabmap standalone version or just ROS?

matlabbe, 2022-03-09:
For pure visual odometry approaches, you could only feed the output odometry topic to rtabmap node in ROS without recompiling rtabmap. 

In the standalone it is maybe more well integrated than using ros version of the packages, but need careful recompiling.

closed 2022-03-11 by nickjrz