introlab/rtabmap #829: Are RTAB-Map binaries available upstream built with libpointmatcher recommended dependencies?

aseligmann, 2022-02-16:
Hello,

Are the RTAB-Map binaries (`ros-noetic-rtabmap`) built with recommended and optional dependencies like libpointmatcher, g2o, GTSAM, non-free OpenCV etc.?

matlabbe, 2022-02-20:
Previous to 0.20.18, rtabmap ros binaries were built with g2o and octomap. From 0.20.18, ros-noetic-rtabmap is also built with libpointmatcher. For melodic, the packages [are not yet released in main repo](http://repositories.ros.org/status_page/melodic_default.html?q=rtabmap) (last blue square is still blue on 0.20.16, wait for next melodic release).

To use GTSAM or non-free OpenCV stuff, we should rebuild from source.

aseligmann, 2022-02-25:
Great, thank you!

closed 2022-02-25 by aseligmann