introlab/rtabmap #831: Confusion with Strategies

EulersBoiler, 2022-02-21:
After reading the paper from 2018, browsing the code and working with RTAB for almost a year, i am still confused with how the strategies for ICP work. 
I do not see the difference between registrationICP which uses the Icp/Strategy parameter to select libpointmatcher, which i assumed to be used for neighbourlink refinment,  and Odom/Strategy used in the construction of odometry objects. Further the Reg/Strategy has been an easy choice to make but i dont know where it influences the flow of data. 

My confusion lately originates from the fact that i can select libpointmatcher and FLOAM at the same time. Thanks for linking sources or a short explanation.

matlabbe, 2022-02-21:
Here is the description of the related parameters (rtabmap --params):
```
Param: Icp/Strategy = "1"  [ICP implementation: 0=Point Cloud Library, 1=libpointmatcher, 2=CCCoreLib (CloudCompare).
Param: Odom/Strategy = "0" [0=Frame-to-Map (F2M) 1=Frame-to-Frame (F2F) 2=Fovis 3=viso2 4=DVO-SLAM 5=ORB_SLAM2 6=OKVIS 7=LOAM 8=MSCKF_VIO 9=VINS-Fusion 10=OpenVINS 11=FLOAM 12=Open3D]
Param: Reg/Strategy = "0"  [0=Vis, 1=Icp, 2=VisIcp]
Param: RGBD/NeighborLinkRefining = "false"  [When a new node is added to the graph, the transformation of its neighbor link to the previous node is refined using registration approach selected (Reg/Strategy).]
```

For odometry node/thread, you have to select `Odom/Strategy` first (default F2M). Then if F2M or F2F is used, `Reg/Strategy` will select which registration approach is used, either visual-based or icp-based transformation estimation. If `Reg/Strategy`>=1, you can then further select which ICP approach is used with `Icp/Strategy` (default libpointmatcher). If `Odom/Strategy` is LOAM, `Reg/Strategy` and `Icp/Strategy` are ignored by the odometry (it is using LOAM directly).

For rtabmap node/thread, for loop closure transformation estimation and neighbor link refining, you have to select `Reg/Strategy` (visual or icp). If `Reg/Strategy`>=1, you can then further select which ICP approach is used with `Icp/Strategy` (default libpointmatcher).

cheers,
Mathieu

EulersBoiler, 2022-02-21:
Really appreciate your help. cheers

closed 2022-02-21 by matlabbe