introlab/rtabmap #832: Visualizer (docked panel and separate window) drags the entire window on pan

parkerlreed, 2022-02-22:
Arch Linux
VTK 9.2
Qt 5.15.2
pcl 1.12.1

CMake flags https://gist.github.com/parkerlreed/09af1075135fd25108a9344607fa0552

Both in the main window docked and the visualize point cloud, dragging the view to pan drags the entire window. Attempting to move with the arrow keys has very bad performance.

https://user-images.githubusercontent.com/841440/155212658-bcf2731d-4348-4210-828f-8c03e9759a21.mp4

parkerlreed, 2022-02-22:
Ok so for Plasma specifically I can disable the dragging anywhere and this indeed keeps it from dragging 

Thing is this option has always been on for me (Drag from empty) and this has never happened before in rtabmap. I would assume those shouldn't be counted as empty areas, and maybe some library updates have messed that up.

[image]

matlabbe, 2022-02-23:
So the whole window is considered an empty area? If it didn't happen before on same system, I think it could be related to Qt version. If you drag on the loop closure detection or odometry views, does it move too? If not, it could be that only OpenGL widgets are detected as empty. If so, the bug could be coming from QVTKOpenGLNativeWidget / QOpenGLWidget on how they capture the mouse events.

parkerlreed, 2022-02-23:
Yep dragging from any of the areas (loop closure/etc) drags the window.

parkerlreed, 2022-02-23:
New observation: I changed the setting back to drag on empty to record the other areas being affected and noticed touch input is not affected in the same way. Disabling touchscreen does not affect the mouse behavior.

https://user-images.githubusercontent.com/841440/155261118-7c020b1f-edc0-466b-bc03-6a8062eba5ce.mp4

parkerlreed, 2023-09-17:
As of Plasma 5.27 (Using the rtabmap jammy container in podman) this is still occurring. Mouse drag is pulling the whole window.

Plasma has seemingly removed the option to change the "inactive" drag behavior, so this is no longer working as a fix.