introlab/rtabmap #833: rtabmap in ROS 2 Rolling has not been released on Ubuntu Jammy

nuclearsandwich, 2022-02-23:
The ROS 2 Rolling Ridley distribution recently [migrated to Ubuntu 22.04 Jammy](https://github.com/ros/rosdistro/pull/32036) using the rosdistro migration tool.

This repository is one which failed to bloom during the migration because the tag convention used in the rtabmap repository was not handled by the migration tool. It should be possible to release this repository in Rolling again by running bloom normally.

Packages which are not released in Rolling may not be automatically added to future stable ROS 2 distributions.

matlabbe, 2022-02-24:
For rolling, some upstream packages that rtabmap depends on were not released on rolling the last time I checked (like nav2 I think). I'll take a look soon to make a new release.

matlabbe, 2022-02-27:
Re-bloomed under https://github.com/ros/rosdistro/pull/32309