introlab/rtabmap #835: No such file or directory  #include <cs.h>

lucamozza, 2022-03-01:
Hi!
I'm getting this error when compiling rtab-map wit g2o
```
/usr/local/include/g2o/solvers/csparse/csparse_extension.h:27:10: fatal error: cs.h: No such file or directory
 #include <cs.h>
```
Apparently, it's the same as #502, however i do have  `libsuitesparse-dev` and `/usr/include/suitesparse/cs.h` does exist.

Any suggestions or debugging steps?

Thanks!

matlabbe, 2022-03-07:
Which g2o version is used? It seems built from source. It happened to me when `g2oConfig.cmake` is used, the include directory of `cs.h` was not added to g2o include directories from installed g2oTargets.cmake:
```
# Create imported target g2o::csparse_extension
add_library(g2o::csparse_extension SHARED IMPORTED)

set_target_properties(g2o::csparse_extension PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "/opt/ros/noetic/include"
)
```
You may try adding `/usr/include/suitesparse` to the interface directories:
`  INTERFACE_INCLUDE_DIRECTORIES "/opt/ros/noetic/include;/usr/include/suitesparse"`

It was handled in old approach to get g2o: https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindG2O.cmake, but now with targets, this issue appeared.

lucamozza, 2022-03-21:
Yes, I made the wrong assumption that I needed to compile g2o myself in my setup.
The problem does not appear when installing from the package manager, I guess it's the same issue you are referring to.
Closing the issue, thanks a lot for getting back to me!

closed 2022-03-21 by lucamozza

iportolesrb, 2022-03-21:
Hi @lucamozza , same problem here, how did you install g2o from package manager? I tried with `sudo apt install ros-melodic-libg2o` but the error persists.

lucamozza, 2022-03-31:
Hi @iportolesrb, sorry for taking so long.
Actually, that's the exact command I used and it worked first try.
Maybe you tried installing it from source previously and it's somehow conflicting? I remember something like that happening to me in the past

iportolesrb, 2022-04-01:
@lucamozza ok, thank you for your answer