introlab/rtabmap #838: rtabmap-export deteriorated

iportolesrb, 2022-03-21:
Hi,

I just updated ros-melodic-rtabmap to 0.20.18 and writing the same command (`rtabmap-export --voxel 0.02 --decimation 4 --max-range 5.0 --mesh rtabmap.db`) I get a mesh significantly worse and smaller than using 0.20.16. When some days ago I got an accurate 600MB file, now I get a fuzzy 10MB one. What was the significant change that makes the map worse? Can I obtain something like in 0.20.16 changing the values of the parameters?

Thank you,
Inés

iportolesrb, 2022-03-22:
Sorry for closing and opening the issue, it was a mistake. 

The problem is not about the version, the first time I used the filter_ceiling filter and I didn't for the second. I don't understand the result because filtering should remove points and be more lightweight, but when I use it is when I get the 600MB map. Maybe for a huge number of points it is not able to get an accurate mesh?

matlabbe, 2022-04-10:
Sorry for the delay, had a paper deadline last week. When the mesh is created, it automatically computes the ideal Poisson reconstruction depth based on the dimension of the point cloud. By filtering the ceiling, you may have remove very far points that caused poor resolution of the mesh. Note that `--max-range` should be `--max_range` (don't think you need to filter the ceiling if the max range is set correctly).
```
rtabmap-export --voxel 0.02 --decimation 4 --max_range 5.0 --mesh rtabmap.db
```

Poisson's depth can be set manually with `--poisson_depth 9`, you may start with 8 or 9, then go up 10 or 11 if you have a large environment.

cheers,
Mathieu

iportolesrb, 2022-04-19:
Hi @matlabbe ,

Ok, I understand, I changed max-range to max_range. Unfortunately, using max_range I have the same problem. I obtain a good map only when I use filter_ceiling with a value less than 2.0. The ceiling is white and there is a lot of noise there, so maybe there are too many bad points that can cause a poor resolution (not only the very far points), so that when I filter the ceiling the mesh can be computed well.

But for some reason it is not the same for the filter_floor, even when there is a lot of noise too because of the brightness, that introduces several reflected points. In this case, if I don't use the filter_floor or the value is over 0.1, I get a good result. Else (filter_floor under 0.1) the map is poor like above.

> Poisson's depth can be set manually with --poisson_depth 9, you may start with 8 or 9, then go up 10 or 11 if you have a large environment.

Also, when I use --poisson_depth the mesh is poor again, no matter what filter_ceiling value I use.

Regards,
Inés

iportolesrb, 2022-05-11:
Hi, how can I share it?

matlabbe, 2022-05-11:
Put it on a dropbox or google drive, share the link here or send me the link on my email (my username here @ gmail) if you want to keep it private.