introlab/rtabmap #842: Get occupancy grid and robot position externally

dequubi, 2022-03-25:
Hi.

I have a question, is there a way to get occupancy grid & robot position updates externally?

Basically I want to use RTAB-Map as a back-end and show the resulting 2D map & robot position in another software *while mapping process is active*.

I need it for my Bachelor thesis and mostly for cosmetic reasons.

matlabbe, 2022-04-11:
This easiest way is to use rtabmap_ros. You can use TF to get the pose at anytime, and the occupancy grid is published on a standard topic.

dequubi, 2022-04-13:
Thank you for your answer.

I've learned a bit about ROS, and I was able to get occupancy grid data through `/rtabmap/grid_map` topic. But now I need to get map's width and height, because `/rtabmap/grid_map` returns an unformatted tuple.

I've found that `MapMetaData` class contains width and height, but I couldn't find a way to get it. What am I missing? 

My current code:

```py
#!/usr/bin/python3
import rospy

from nav_msgs.msg import OccupancyGrid

def callback(data):
    occupGrid = data.data
    rospy.loginfo(rospy.get_caller_id() + ": I heard %s", occupGrid)

def listener():
    rospy.init_node('listener', anonymous=True)

    rospy.Subscriber("/rtabmap/grid_map", OccupancyGrid, callback)

    rospy.spin()

if __name__ == '__main__':
    listener()
```

It returns:

[image]

(and a few thousand numbers more)

stevemartinov, 2022-04-17:
This is the message type:
http://docs.ros.org/en/lunar/api/nav_msgs/html/msg/OccupancyGrid.html

You are getting data, not info. Info is MapMetaData type which has what you need. 

`occupGrid = data.info`

dequubi, 2022-04-20:
I have no idea how I missed that, thank you very much (and I do need both data and info).

I struggled a lot with tf lookups, but finally got everything working.

I will close this issue once I'm sure I have no questions left.