introlab/rtabmap #844: How to optimize existing coarse poses with rabmap loop closure detection ?

MemoryNode, 2022-04-04:
with other slam front-end( like vins-mono or orb-slam ...),  i got the rgbd(keyframe)+poses ( synchronized ),  but there are some loop closure errors, i want to optimize the existing coarse poses with rtabmap's loop closure detedtion for mapping, but i don't know how to do, is there a similar tutorial ?

Thank you very much in advance for your help.

rlabs-oss, 2022-04-04:
Use stop if currently mapping, or load the .db and use Tools -> post processing / Detect more loop closures.

MemoryNode, 2022-04-05:
> Use stop if currently mapping, or load the .db and use Tools -> post processing / Detect more loop closures.

Thanks for your reply @rlabs-oss ,  what i have are some offline file (depth_imgs / rgb_imgs / camera_info.yaml / coarse_pose.txt(with format:  timestamp x y z qx qy qz qw) ).
[image]

 i followed the noEventsExample code, change the CameraStereoImages to CameraRGBDImages, and change rtabmap.process(data, pose) to rtabmap.process(data,coarse_pose), but i only got a messy result.
[image]

i have no db file, and how to feed the all offline data to rtabmap, and get the optimized map, i'm in trouble... -_- !!!

rlabs-oss, 2022-04-05:
For the stand-alone app, In preferences set the driver to RGB-D, and choose images as the driver, set the path for RGB and Depth images and odometry file to coarse_pose_sync.txt - the pose needs to be in a format present in the drop down menu, e.g RGB-D-SLAM. Hover over the Odometry format to see the tool tip describing each format. Then new database and start.

MemoryNode, 2022-04-05:
Hi, @rlabs-oss

I tried following the process you said, but I still got a messy map(as the left).  Then, i tried to map with the coarse pose and depth,  and from the result, it was correct(as the right).
[image]

The coarse poses are in the right-hand coordinate with RGBD-SLAM pose format,  i also tried to convert them to left-hand, didn't work...

rlabs-oss, 2022-04-05:
Looks good, the pose you choose in RTAB-Map needs to match the format you used while capturing or the capturing app if external needs to write the pose in the format needed (or convert an existing one if that's not possible) -if you omit the pose file entirely and start mapping with the RGB-D files (image and depth) does the result look the same as the viewer image (on the right) ?

MemoryNode, 2022-04-06:
hi, @rlabs-oss 

a little good things. if i start without odometry file, RTAB-MAP will ues it's own odometry, and this will result in the loss of odometry(red background), then i tried set the (coarse pose)odometry file with RGBD-SLAM(motion capture) format, it looks good, the 3d map is the same as the 'mapping with coarse pose' .
[image]

but what's the 'motion capture' mean ? and what 's the difference between RGBD-SALM and RGBD-SLAM(motion capture) format ?

MemoryNode, 2022-04-08:
Thanks for your help @rlabs-oss , I can continue to do the follow-up test .

closed 2022-04-08 by MemoryNode

matlabbe, 2022-04-11:
Hi, 

motion capture format is the format used by [TUM's RGB-D dataset](https://vision.in.tum.de/data/datasets/rgbd-dataset) (don't remember exactly which orientation it is). The one without "motion capture" is ROS format (x->forward, y->left, z->up).

By curiosity, your depth image seems pretty dense in texture-less areas, which sensor are you using?

cheers,
Mathieu

MemoryNode, 2022-04-17:
Hi, @matlabbe 

Thank you for your explanation of motion capture format,  the rgb and depth are captured on an iPhone/iPad.

cdb0y511, 2022-04-18:
> Hi,
> 
> motion capture format is the format used by [TUM's RGB-D dataset](https://vision.in.tum.de/data/datasets/rgbd-dataset) for their ground truth (don't remember exactly which orientation it is, probably camera frame x->right, y->down, z->forward). The one without "motion capture" is ROS format (x->forward, y->left, z->up).
> 
> By curiosity, your depth image seems pretty dense in texture-less areas, which sensor are you using?
> 
> cheers, Mathieu

Hi, @matlabbe. I meet the same issue too. It confused me for a long time.
I want to load my trajectories into the standalone version. But I notice it may use a different coordinate system. The result with format motion capture is similar to the above (disordered).
I think the orientation of my coordinate system is camera frame x->right, y->down, z->forward. I think the camera frame is related to the back-projection process (2.5D to 3D, correct me if I was wrong), top left of the depth map is the original. The back-projection is similar to rtabmap, but I notice rtabmap adds a local transform to the back-projected 3D point, i.e.,
( 0  0  1  0
-1  0  0  0
 0 -1  0  0
 0  0  0  1) 
In the graph.CPP says remove optical rotation, and z pointing front, x left, y down. (I think this is the part transforming the different coordinate system to rtabmap)
https://github.com/introlab/rtabmap/blob/f140e99881a6fd0b6c81d5a29a13fceee28cf2f1/corelib/src/Graph.cpp#L457
https://github.com/introlab/rtabmap/blob/f140e99881a6fd0b6c81d5a29a13fceee28cf2f1/corelib/src/Graph.cpp#L481
I can understand this part
					Transform t( 0, 0, 1, 0,
								-1, 0, 0, 0,
								 0,-1, 0, 0);
					pose = t * pose * t.inverse();
But why another rotation?
							t = Transform( 0, 0, 1, 0,
										   0, -1, 0, 0,
										   1, 0, 0, 0);
							pose = t*pose;
In all, how to transform the trajectories to rtabmap coordinate system?

matlabbe, 2022-05-09:
Note that you can download [RTAB-Map on AppStore](https://apps.apple.com/ae/app/rtab-map-3d-lidar-scanner/id1564774365) to create db directly on iPhone/iPad.