introlab/rtabmap #847: If it is possible to add function like manual key point select and association in database editor?

wystephen, 2022-04-09:
It is hard to achieve multi-session mapping if the environment is complex. Suppose it is possible to add point association manually. The method will be more robust and provide better results.

stevemartinov, 2022-04-17:
Disagree completely. If you are using this for a project with 1 robot then may be but many people use it in large commercial scale setups where they cannot manually add something when thousands of robots are deployed all over the world

wystephen, 2022-04-18:
It is impossible to add an association manual for the whole dataset, but it is possible and valuable to add association at some keyframe.
It is hard to avoid some time keyframe taken at same place not correctly associated. If it is possible to manually add association rather than re-collect a new dataset to connect these two frames can significantly reduce human resources requirements.

matlabbe, 2022-05-09:
From what I understand of the original issue "manual key point select and association in database editor", so you mean instead of using OpenCV keypoint detector to extract keypoints and find correspondences, to manually click on same pixels from two different frame (similar to some Structure-From-Motion pieplines requiring the user to manually click on corresponding pixels). Unless you have a very repetitive pattern that you need to match (like a checker board seen by two frames), personally by changing the detector used, number of features extracted, lower the min inliers theshold, I am generally to find enough corresponding features automatically, using Constraints View in DatabaseViewer. You can open the Core Parameters panel, go under Vis group, then modify parameters like you want. You may have to check also `RGBD/LoopClosureReextractFeatures` parameters under Core Parameters->RGBD group if you change the detector used (look at the terminal for warnings if adding new loop closure is rejected or a detector is not available because rtabmap is not built with it). Respective parameters for each detector are under the group with same name
[image]
.

wystephen, 2022-05-10:
Thanks for your advice. The easiest way might be to collect a short-term dataset around an area that cannot detect loop closure successfully.