introlab/rtabmap #849: Projection matrix

luantunez, 2022-04-12:
Hello! I am not being able to obtain the projection matrix from the desktop app using a iOS scan with LiDAR. Can you help to obtain that data, together with the intrinsics and camera pose matrices?

Thank you very much in advance!

stevemartinov, 2022-04-17:
Before asking this I would suggest to post what you have tried to acheive yourself, what where the results and where exactly you are stuck and need help. You need to search/work yourself first

luantunez, 2022-04-17:
I have been trying to get the projection matrix from your code extracting it from the ARframe, as an option, but it does not work for projecting the texture to the point cloud. The thing is that from you code I see that the output calibration.yaml file can give a projection matrix together with the camera matrix, intrinsics, distortion coefficients, etc. But with my LiDAR camera from the iPhone 12 pro I am not getting all of that information ( and  specifically, the projection matrix I am looking for). I was wondering if there is a way of getting that more complete yaml file.
Thank you again!