introlab/rtabmap #850: When my robot is stationary, the relocation will still fluctuate.

lamer-afk, 2022-04-13:
When my robot is stationary, the relocation will still fluctuate. 
What is the reason?

By the way, what's the difference between a Local Loop closure and a global loop closure?

stevemartinov, 2022-04-17:
It depends on your parameters. If you have enabled `ProximityBySpace` then very possible that rtabmap tries to find local matches if your robot is near to the keyframes. 

You should read the paper first: https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf

It is all written there

lamer-afk, 2022-04-21:
@stevemartinov  thanks a lot
By the way, how do you visually see which key points match when relocating?
When I use the Rtabmap-DatabaseViewer, it seems to regenerate the key points rather than recording the key points at the time.