introlab/rtabmap #854: Superpoint crashing RTAB-Map

pVenkataa, 2022-04-21:
I am using RTABMAP on Linux whenever I am selecting the SuperPoint feature detector, and Rtabmap crashes.
It shows the following error : 
terminate called after throwing an instance of 'c10::Error'
  what():  Legacy model format is not supported on mobile.
Exception raised from deserialize at ../torch/csrc/jit/serialization/import.cpp:267 (most recent call first):

Linux: 18.04
Libtorch: https://download.pytorch.org/libtorch/cpu/libtorch-cxx11-abi-shared-with-deps-1.11.0%2Bcpu.zip
SuperPoint model weights: https://github.com/KinglittleQ/SuperPoint_SLAM Superpoint.pt file from this link.
[image]

hellovuong, 2022-04-24:
Because you are using incorrect libtorch version, I met same problem and change to libtorch1.4 solved it.

pVenkataa, 2022-04-24:
Solved the error by installing Libtorch LTS (1.8.2) version.

closed 2022-04-24 by pVenkataa

mattiasmar, 2023-06-18:
@venkataPalacherla  @matlabbe I'm running RTabMap within a ubuntu22 docker. I've installed PyTorch 2.0.1+cu117, and my host machine has Cuda 12.1 installed. Is this expected to break SuperPoint?

I get similar errors as Venkata:
[image]

matlabbe, 2023-06-20:
Error is "`PytorchStreamReader failed reading zip archive: failed finding central directory`", not sure what it means, it looks pytorch related. You may have to regenerate `Superpoint.pt` file that is compatible with your pytorch version, see "Generate `superpoint_v1.pt` .. " [here](https://github.com/introlab/rtabmap/tree/master/archive/2022-IlluminationInvariant#how-reproduce-results-shown-in-the-paper)