introlab/rtabmap #861: Win64 with Cuda11

rlabs-oss, 2022-05-07:
New Windows laptop:

- Installed realsense binary, and rtabmap 0.20.16-win64-cuda11-1.exe - on startup "Missing MSVCR100.dll" error *1, the download on the microsoft site is 404 for that library. 
re: [Install visual studio 2010 redistributable](https://www.microsoft.com/en-US/Download/details.aspx?id=14632) if you have "Missing MSVCR100.dll".

*1Followed by "boost-chrono-vc140-mt-1_62.dll not found", opencv_cudaimgproc450.dll etc

Visual Studio 2022 is installed. 

also tried the non-cuda release which returned same MSVCR100.dll error. 

Install was "for all users"

rlabs-oss, 2022-05-07:
Solved, update the URL for MSVCR100.dll to the service pack 1 release

See:-
https://www.microsoft.com/en-us/download/details.aspx?id=26999

closed 2022-05-07 by rlabs-oss

matlabbe, 2022-05-09:
Thx for the updated link. I modified the one  here https://github.com/introlab/rtabmap/wiki/Installation#windows