introlab/rtabmap #866: RGB camera calibration hangs

ckiee, 2022-05-17:
I went into the calibration GUI, downloaded the PDF, displayed it on a secondary monitor and moved the webcam around it from different angles/distances until the count progress bar finished at around 42. I left it to process but poking it with `gdb` has revealed it's still doing the exact same thing as yesterday:

```
(gdb) where
#0  0x00007fa4e9067567 in sched_yield () from /nix/store/m4g6lswi75b739cpdx8wfxlfmcazyks9-glibc-2.34-115/lib/libc.so.6
#1  0x00007fa4f7ee278d in inner_thread () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#2  0x00007fa4f8016153 in exec_blas._omp_fn () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#3  0x00007fa4f9db1a86 in GOMP_parallel () from /nix/store/69brclzxp7mg927k6986hrfzyd1hpqgd-gcc-11.2.0-lib/lib/libgomp.so.1
#4  0x00007fa4f801656f in exec_blas () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#5  0x00007fa4f7ee2d53 in gemm_driver.isra () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#6  0x00007fa4f7ee2e9b in dgemm_thread_nt () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#7  0x00007fa4f7df3b07 in dgemm_ () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#8  0x00007fa4f8ee0a49 in dgebrd_ () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#9  0x00007fa4f8ef22f9 in dgesdd_ () from /nix/store/j43l6bmil57qmr4pv8srl3xmc2cdc7px-blas-3/lib/libblas.so.3
#10 0x00007fa4f47de418 in lapack_SVD64f(double*, unsigned long, double*, double*, unsigned long, double*, unsigned long, int, int, int) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_core.so.4.5
#11 0x00007fa4f47f5469 in cv::hal::SVD64f(double*, unsigned long, double*, double*, unsigned long, double*, unsigned long, int, int, int) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_core.so.4.5
#12 0x00007fa4f47f5d89 in cv::solve(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_core.so.4.5
#13 0x00007fa4f52b9fb9 in CvLevMarq::step() () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_calib3d.so.4.5
#14 0x00007fa4f52ba82f in CvLevMarq::updateAlt(CvMat const*&, CvMat*&, CvMat*&, double*&) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_calib3d.so.4.5
#15 0x00007fa4f526cb64 in cvCalibrateCamera2Internal(CvMat const*, CvMat const*, CvMat const*, CvSize, int, CvMat*, CvMat*, CvMat*, CvMat*, CvMat*, CvMat*, CvMat*, int, CvTermCriteria) ()
   from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_calib3d.so.4.5
#16 0x00007fa4f527067b in cv::calibrateCameraRO(cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, int, cv::_InputOutputArray const&, cv::_InputOutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, int, cv::TermCriteria) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_calib3d.so.4.5
#17 0x00007fa4f5271039 in cv::calibrateCamera(cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, cv::_InputOutputArray const&, cv::_InputOutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, int, cv::TermCriteria) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_calib3d.so.4.5
#18 0x00007fa4f527112a in cv::calibrateCamera(cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, cv::_InputOutputArray const&, cv::_InputOutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, int, cv::TermCriteria) () from /nix/store/3p96pgq1lkhirs69qc0r1687lzaccb2n-opencv-4.5.4/lib/libopencv_calib3d.so.4.5
#19 0x00007fa4fadb6ca3 in rtabmap::CalibrationDialog::calibrate() () from /nix/store/ippf9zdmw9yqa6wv006za12sa3k17yq3-rtabmap-unstable-2022-02-07/lib/librtabmap_gui.so.0.20
#20 0x00007fa4e97ed618 in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#21 0x00007fa4ea36e722 in QAbstractButton::clicked(bool) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#22 0x00007fa4ea36e99a in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#23 0x00007fa4ea370560 in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#24 0x00007fa4ea370783 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#25 0x00007fa4ea2bc70e in QWidget::event(QEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#26 0x00007fa4ea27a6cf in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#27 0x00007fa4ea282224 in QApplication::notify(QObject*, QEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#28 0x00007fa4e97b6a2a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#29 0x00007fa4ea280d17 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#30 0x00007fa4ea2d583c in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#31 0x00007fa4ea2d8ab5 in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#32 0x00007fa4ea27a6cf in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#33 0x00007fa4e97b6a2a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#34 0x00007fa4e9bc87b7 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Gui.so.5
#35 0x00007fa4e9b9e56c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Gui.so.5
#36 0x00007fa46eeeaeca in ?? () from /nix/store/vdfcc2c8ndvhfcx5xjjwhbn95cpxj457-qtbase-5.15.3-bin/lib/qt-5.15.3/plugins/platforms/../../../../../5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5XcbQpa.so.5
#37 0x00007fa4e65d5ddb in g_main_context_dispatch () from /nix/store/hqvf17nn5fmwrlrgq9qyplz1id923jri-glib-2.72.1/lib/libglib-2.0.so.0
#38 0x00007fa4e65d6088 in g_main_context_iterate.constprop () from /nix/store/hqvf17nn5fmwrlrgq9qyplz1id923jri-glib-2.72.1/lib/libglib-2.0.so.0
#39 0x00007fa4e65d613f in g_main_context_iteration () from /nix/store/hqvf17nn5fmwrlrgq9qyplz1id923jri-glib-2.72.1/lib/libglib-2.0.so.0
#40 0x00007fa4e980e494 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#41 0x00007fa4e97b53ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#42 0x00007fa4ea485cba in QDialog::exec() () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#43 0x00007fa4fabe0189 in rtabmap::PreferencesDialog::calibrate() () from /nix/store/ippf9zdmw9yqa6wv006za12sa3k17yq3-rtabmap-unstable-2022-02-07/lib/librtabmap_gui.so.0.20
#44 0x00007fa4e97ed618 in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#45 0x00007fa4ea36e722 in QAbstractButton::clicked(bool) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#46 0x00007fa4ea36e99a in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
#47 0x00007fa4ea370560 in ?? () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Widgets.so.5
[... cut ...]
#87 0x00007fa4e980e494 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#88 0x00007fa4e97b53ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#89 0x00007fa4e97bd9f0 in QCoreApplication::exec() () from /nix/store/5zdacvyjfx4gg1lyfj7kfr3h781jqf7q-qtbase-5.15.3/lib/libQt5Core.so.5
#90 0x00000000004036d5 in main ()
```

Do you think it is actually working on it or is it stuck? How can I debug this or add some progress indicator?

koshiba2021, 2022-06-15:
私も同じ問題を抱えています
キャリブレーションが終了しません
[image]

ckiee, 2022-06-15:
Try selecting *Fish eye* since your camera looks like that. Then you press the *Calibrate* button after a while.

koshiba2021, 2022-06-15:
Thank you for your reply.
I have successfully calibrated !!
Then I generated a cam.yaml file.

I immediately tried localization using that file.
But I got the following error:

-------------------
[ERROR] (2022-06-16 01:46:25) OdometryThread.cpp:140::rtabmap::OdometryThread::addData() Missing some information (images/scans empty or missing calibration)!?
-------------------

Is localization impossible with a cheap USB webcam that doesn't have an odometry sensor
[image]
?

ckiee, 2022-06-15:
Maybe you have to save it first, I'm not very sure though, I was planning to map with a kinect (RGBD) and then use a webcam (just RGB) with it, but I haven't finished that project.

koshiba2021, 2022-06-15:
Thank you for your reply.

One person creates a map (map.db) using Azure Kinect / Realsense.
Other users can estimate their own position with the above map. Using only one cheap USB WebCAM (RGB).

It would be great if this were possible.
I searched on Google search but couldn't find any software to enable it.

I will try to study ORB-SLAM.

matlabbe, 2022-10-09:
There is no mono odometry approach officially implemented in rtabmap. If you only want to localize with RGB-only camera in pre-built map, there is a specific option in Preferences to not start odometry:
[image]

Kyu, 2023-04-01:
> There is no mono odometry approach officially implemented in rtabmap. If you only want to localize with RGB-only camera in pre-built map, there is a specific option in Preferences to not start odometry: [image]

Note: for this setting to be enabled, you must first disable RBG-D Slam using Advanced Settings, as seen on the left panel

matlabbe, 2023-04-02:
I think this is this one you are referring:
[image]

matlabbe, 2023-08-27:
Missing calibration? Did you pass a YAML file along your video in Source panel?

fbliman, 2023-08-28:
Yes, I when I hit the test button I see the video correctly, but then when I start the localization i get that error

matlabbe, 2023-08-29:
The video preview doesn't do any checks about calibration. 
[image]

Oh just realized the warning was in Odometry, rtabmap's odometry doesn't support monocular images. If you want to test purely localization, you can disable odometry with this option:
[image]