introlab/rtabmap #868: [BUILD ERROR] undefined reference to `g2o:: ...

adbidwai, 2022-05-23:
**System Config:**
OS : Ubuntu 18.04
ROS Distro : Melodic

**Issue:**
I am trying to install `rtabmap` from source. I have installed g2o from source as well. Getting the following error when I am trying to build `rtabmap` from source.

**Error:**
```
$USER@Cadence:~/nus_ws/src/rtabmap/build$ make
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[ 43%] Built target rtabmap_core
[ 80%] Built target rtabmap_gui
Consolidate compiler generated dependencies of target rtabmap
[ 80%] Linking CXX executable ../../bin/rtabmap
../../bin/librtabmap_core.so.0.20.16: undefined reference to `g2o::OptimizableGraph::Edge::clone() const'
../../bin/librtabmap_core.so.0.20.16: undefined reference to `g2o::OptimizableGraph::Vertex::clone() const'
../../bin/librtabmap_core.so.0.20.16: undefined reference to `g2o::VertexSBAPointXYZ::VertexSBAPointXYZ()'
../../bin/librtabmap_core.so.0.20.16: undefined reference to `g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*)'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:494: recipe for target 'bin/rtabmap' failed
make[2]: *** [bin/rtabmap] Error 1
CMakeFiles/Makefile2:880: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
```

sumitsarkar1, 2022-05-23:
Try this commit `9b41a4e` for g2o

adbidwai, 2022-05-23:
@sumitsarkar1 checking out to the commit you mentioned, worked. Thanks !!

closed 2022-05-23 by adbidwai