introlab/rtabmap #869: RTABMAP use images directory with problem that no camear model is loaded or valid.

wystephen, 2022-05-24:
# Error Info
[ERROR] (2022-05-24 12:55:49.053) CameraImages.cpp:225::init() Parameter "rectifyImages" is set, but no camera model is loaded or valid.

# Hyperparameters
Source type RGB-> Images.
The calibration file path is given and generated based on [creating  calibration] button.

realyyz, 2022-05-25:
Try version 0.18.0 instead the latest one. I had the same problem with the latest version but I am OK with version 0.18.0.

wystephen, 2022-05-25:
Thanks for your advise. I'll try it.
But this problem might be a bug in the system.

AbhiKhoyani, 2022-07-21:
Hi @wystephen  Did you find any solution for this problem other than downgrading to version 0.18.0?

matlabbe, 2022-10-06:
I tested with this [dataset](https://github.com/introlab/rtabmap/wiki/Stereo-mapping#process-a-directory-of-stereo-images) in RGB and stereo mode, and I cannot reproduce the problem on latest version on master. I know I fixed [some stuff](https://github.com/introlab/rtabmap/commit/83c1adfd1e2ddcee2c4d97c6f69f7dcd04903e6e) about CameraImages this summer, maybe this fixed that issue. 

Note that for the RGB-only mode, it won't work in metric SLAM mode. This option is there just for the good old days (original version which was only appearance-based mapping like the name still say).

If you can share samples of your dataset, I could try on my side and fix the bug if there is one.