introlab/rtabmap #872: rtabmap: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

MymengA, 2022-05-29:
Hello everyone, I got the trouble when I run the Rtab with ORB3 on ROS noteic with Opencv4

my environment:
1.I installed the ROS noteic
2.I 
3.I just run
`roslaunch rtabmap_ros euroc_datasets.launch args:="Odom/Strategy 5 OdomORBSLAM/VocPath /home/meng/rtab_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt" MH_seq:=true raw_images_for_odom:=true`

`rosbag play MH_01_easy.bag`

I error is
`rtabmap: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
`

hariharan382, 2022-07-19:
Any idea on how you solved it

matlabbe, 2022-10-10:
You may try this: https://github.com/introlab/rtabmap/issues/655#issuecomment-851037772

closed 2024-03-31 by matlabbe