introlab/rtabmap #873: Error in transforming GPS bearing

kurtgeebelen, 2022-05-30:
Hi Matthieu,

I think the transformation on the bearing you do for GPS has a small mistake:
https://github.com/introlab/rtabmap/blob/8f8256c1dd60eceb5daea8cb4ffa0aa24cf57578/corelib/src/Memory.cpp#L5817

I I understand correctly, x-axis is east and y-axis is north, with 0 degrees of bearing being north and clockwise, and data.gps().bearing() is in degrees, correct? So you subtract 90 degrees to make x-axis 0 degrees, and take negative to make it anticlockwise. 

But I guess you would then need to do *M_PI/180.0  (and not *180.0/M_PI) to go to radians?

Best,
Kurt

kurtgeebelen, 2022-05-30:
I just see that you put the InfMatrix for orientation (so also for yaw) >9999, so it would get ignored in the optimizer and would not have caused problems so far, but not sure why the orientation from GPS is not used?

matlabbe, 2022-09-06:
Indeed, it seems an error. Those other places are like you suggested: 
https://github.com/introlab/rtabmap/blob/f1c987a0cefd183efe93f9521ee0656440126918/corelib/src/Rtabmap.cpp#L1810

https://github.com/introlab/rtabmap/blob/aa3b71dbf663d884a0293da7eff7444064e341e1/guilib/src/MainWindow.cpp#L2360

https://github.com/introlab/rtabmap/blob/aa3b71dbf663d884a0293da7eff7444064e341e1/guilib/src/DatabaseViewer.cpp#L1884

The bearing value is not always available (depending on the phone/GPS hardware), so it is why it is not used in graph optimization. I'll fix the one you pointed out.

commit 0dc5fbdd (2022-09-06): fixed #873
closed 2022-09-06 by matlabbe