introlab/rtabmap #876: getMemoryUsed() CV_DbgAssert

triplehoon, 2022-07-01:
I am using Rtabmap with openCV 4.5.

I got an assertion when I try to run in debug mode.

Especially, in the function getMemoryUsed() in SensorData.cpp.

When calculating elemSize, the openCV assertion came out

```cpp
inline
size_t Mat::elemSize() const
{
    size_t res = dims > 0 ? step.p[dims - 1] : 0;
   CV_DbgAssert(res != 0); // <<< assertion in here
    return res;
}
```

Is it by design? Or openCV version problem?

matlabbe, 2022-09-14:
Not sure, it probably means we are calling `elemSize()` on an empty `cv::Mat()`. Looking at this code:
https://github.com/introlab/rtabmap/blob/aa3b71dbf663d884a0293da7eff7444064e341e1/corelib/src/SensorData.cpp#L812-L829
that is probably true! We would have to check `cv::Mat::empty()?` before calling `elemSize()`... Other locations where `elemSize()` is used: https://github.com/introlab/rtabmap/search?q=elemSize

matlabbe, 2023-09-17:
Fixed by https://github.com/introlab/rtabmap/commit/f1cd819673804e0c5beaad1f5b316b12c4dcf140

closed 2023-09-17 by matlabbe