introlab/rtabmap #879: Disable saving database in localization mode

SamAlexandrov, 2022-07-13:
Hi,
I have similar issue to #342 and what I can see is that the issue was closed without a solution for pure localization. 

In a nutshell, I'm trying to localize on an existing map with the following parameters set according to the documentation:
```
 * /rtabmap/Grid/CellSize: 0.05
 * /rtabmap/Grid/Sensor: 0
 * /rtabmap/Icp/MaxCorrespondenceDistance: 3.0
 * /rtabmap/Icp/MaxRotation: 0.78
 * /rtabmap/Icp/MaxTranslation: 3.0
 * /rtabmap/Icp/VoxelSize: 0
 * /rtabmap/Kp/MaxFeatures: -1
 * /rtabmap/Mem/IncrementalMemory: false
 * /rtabmap/Mem/InitWMWithAllNodes: true
 * /rtabmap/Mem/LocalizationDataSaved: false
 * /rtabmap/Odom/Holonomic: True
 * /rtabmap/OdomF2M/BundleAdjustment: 0
 * /rtabmap/RGBD/ProximityPathMaxNeighbors: 10
 * /rtabmap/Reg/Strategy: 1
 * /rtabmap/Rtabmap/DetectionRate: 0
 * /rtabmap/approx_sync: True
 * /rtabmap/database_path: ~/ros/noetic/syst...
 * /rtabmap/frame_id: base_link
 * /rtabmap/publish_tf: True
 * /rtabmap/scan_normal_k: 5
 * /rtabmap/subscribe_depth: False
 * /rtabmap/subscribe_rgb: False
 * /rtabmap/subscribe_scan: True
 * /rtabmap/subscribe_scan_cloud: False
 * /rtabmap/subscribe_stereo: False
 * /rtabmap/wait_for_transform: True
 * /rtabmap/wait_for_transform_duration: 8
 * /rtabmap/wait_imu_to_init: False
 * /twist_mux/locks: [{'name': 'naviga...
 * /twist_mux/topics: [{'name': 'zero_v...
 * /use_sim_time: True
```

I've disabled the "--delete_db_on_start" and I'm loading successfully my map from a specified directory, using the parameter:
```
<arg name="database_path"    default="~/ros/noetic/system/src/test/sim_map.db"/>
```

However, on exit, I still get the following message:
```
rtabmap: Saving database/long-term memory... (located at /home/sam/ros/noetic/system/src/test/sim_map.db)
rtabmap: 2D occupancy grid map saved.
rtabmap: Saving database/long-term memory...done! (located at /home/sam/ros/noetic/system/src/test/sim_map.db, 380 MB)
```

and the original file is updated and overwritten, similar to the slam case. As I want to load a map, setting the `database_path` to "" is not an option in my case. Does someone have an idea how to set this?

SamAlexandrov, 2022-07-21:
Another question arose in the last couple of days: is it possible to compare two maps (for example the original, and the overwritten maps), such that it can be seen what is changed in the new one? I was looking into the tools like rtabmap-databaseViewer, but all I could find was related to viewing and analysing a single db file :)

matlabbe, 2022-11-26:
Because of the memory management approach of rtabmap, even in localization mode the map could change depending on the nodes transferred to or retrieved from the long-term memory. Same effect if there is more than 1 disjoint session in the database. If you are not using memory management, the map saved back in the database should be the same. 

Note also that the last localization pose is also saved back to database, so that the robot knows where it shut down in the environment (assuming you don't move the robot while it is offline).

To compare maps, you may try just exporting the pgm and compare them in an image editor.