introlab/rtabmap #881: Is it possibe to send the position data from rtab map to other programs ?

TheBricktop, 2022-07-17:
Id like to retrieve positional data from rtabmap using some sort of udp or osc protocol for my interactive installation. Is it currently possible? What should be done to achieve it?

sumitsarkar1, 2022-07-26:
you can use a QUDPSocket class to achieve this...or create a class inheriting from UEventsHandler , register to UEventsHandler and listen to OdometryEvent [like here](https://github.com/introlab/rtabmap/blob/c5e4d67f8003b4a46aeac6e46c9db4000ae24c29/corelib/include/rtabmap/core/OdometryEvent.h#L71)

matlabbe, 2022-07-27:
If you want to do it without ROS, you will have to implement and approach like @sumitsarkar1 . 

Otherwise, if you use ROS, everything is published on topics.

TheBricktop, 2022-08-03:
So basically id need to build rtabmap with additional feature, or You are suggesting using rtabmap as a library?

camasmie, 2022-11-25:
Hi @TheBricktop did you find a solution to get pose information from rtabmap in realtime? We need the same functionality with rtabmap standalone version(non-ROS) and wonder if you used any of @sumitsarkar1 suggestions or something else. More details would be appreciated.

sumitsarkar1, 2022-11-26:
thiis is a hack [https://github.com/sumitsarkar1/rtabmap](url) ...check the guilib folder....better to register to Events...

sumitsarkar1, 2022-11-26:
https://github.com/sumitsarkar1/rtabmap

matlabbe, 2022-11-27:
> So basically id need to build rtabmap with additional feature, or You are suggesting using rtabmap as a library?

That feature is ROS or ROS2 :) Otherwise, you can look at the threaded C++ example [here](https://github.com/introlab/rtabmap/tree/master/examples/RGBDMapping) where the data is shared using events. The idea is when you do a class inheriting from [UEventHandler](UEventHandler), you can see all rtabmap events published.