introlab/rtabmap #886: ORB_SLAM3::Tracking::Tracking no matching function call when building RTABMap

Srijal97, 2022-08-07:
I am trying to build RTABMap (from the noetic-devel branch, and also the latest) with ORB_SLAM3 support on Ubuntu 20. It seems that there is a mismatch at this point between `/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp` and `ORB_SLAM3/include/Tracking.h`. 

I build RTABMap with the following cmake options: `cmake .. -DWITH_G2O=OFF -DWITH_QT=OFF -DWITH_ORB_SLAM=ON`, and here is the output:

```
-- MOBILE_BUILD=OFF
CMake Deprecation Warning at CMakeLists.txt:72 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE  found components: C CXX 
-- Found OpenMP: /usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/include/pcl-1.10;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-7.1;/usr/include/freetype2;/usr/include/x86_64-linux-gnu
-- Found ZLIB: /usr/include
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- GTSAM include directory:  /usr/lib/cmake/GTSAM/../../../include
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
[... 106 lines cut ...]
/root/ORB_SLAM3/include/Tracking.h:61:5: note:   candidate expects 10 arguments, 8 provided
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp: In member function 'void ORB_SLAM3::Tracker::Track()':
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:122:31: error: 'using SE3f = class Sophus::SE3<float>' {aka 'class Sophus::SE3<float>'} has no member named 'empty'
  122 |                  if(mVelocity.empty() || mCurrentFrame.mnId<mnLastRelocFrameId+2)
      |                               ^~~~~
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:156:36: error: 'using SE3f = class Sophus::SE3<float>' {aka 'class Sophus::SE3<float>'} has no member named 'empty'
  156 |                      if(!mVelocity.empty())
      |                                    ^~~~~
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:178:36: error: 'using SE3f = class Sophus::SE3<float>' {aka 'class Sophus::SE3<float>'} has no member named 'empty'
  178 |                      if(!mVelocity.empty())
      |                                    ^~~~~
/root/rtabmap/corelib/src/odometry/OdometryORBSLAM.cpp:183:48: error: 'Sophus::SE3<float> ORB_SLAM3::Frame::mTcw' is private within this context
  183 |                          TcwMM = mCurrentFrame.mTcw.clone();
      |                                                ^~~~
```
and some more. I can add the whole log if it helps. But one of the issues as I see it is that the constructor in `ORB_SLAM3/include/Tracking.h`, `line 61`:
```cpp
Tracking(System* pSys, ORBVocabulary* pVoc, FrameDrawer* pFrameDrawer, MapDrawer* pMapDrawer, Atlas* pAtlas,
             KeyFrameDatabase* pKFDB, const string &strSettingPath, const int sensor, Settings* settings, const string &_nameSeq=std::string());
```
has the additional `Settings* settings` argument which is not taken care of by the call in `OdometryORBSLAM.cpp`, `line 60`:
```cpp
Tracking(pSys, pVoc, pFrameDrawer, pMapDrawer, pMap, pKFDB, strSettingPath, sensor)
```
Any advice on how I could resolve this?

Srijal97, 2022-08-08:
To bypass this error, I am using ORB-SLAM3 `v0.4-beta` for now, and it builds and works just fine. Support for ORB-SLAM3 `v1.0-release` needs to be implemented. 

Another thing, is there a way to change these camera parameters while rtabmap_ros starts up?
```
Camera Parameters: 
- Camera: Pinhole
- fx: 264.699
- fy: 264.699
- cx: 328.622
- cy: 185.428
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: 0
- fps: 30
- color order: RGB (ignored if grayscale)
```
I have the calibration file from my ZED 2i, and can't seem to figure out how I should change these in RTABMap/ORB.

Srijal97, 2022-08-08:
**Accidentally closed the issue, sorry!**

To bypass this error, I am using ORB-SLAM3 v0.4-beta for now, and it builds and works just fine. Support for ORB-SLAM3 v1.0-release needs to be implemented.

Another thing, is there a way to change these camera parameters while rtabmap_ros starts up?

Camera Parameters: 
- Camera: Pinhole
- fx: 264.699
- fy: 264.699
- cx: 328.622
- cy: 185.428
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: 0
- fps: 30
- color order: RGB (ignored if grayscale)

I have the calibration file from my ZED 2i, and can't seem to figure out how I should change these in RTABMap/ORB.

matlabbe, 2022-11-07:
Stick to https://github.com/UZ-SLAMLab/ORB_SLAM3/releases/tag/v0.4-beta with this patch https://gist.github.com/matlabbe/f5cb281304a1305b2824a6ce19792e13 for now. They did a lot of API breaking changes between v0.4 and v1.0. Would need to take 1 day or two to make it work with v1.0.

PonyPC, 2023-03-05:
Is there any update about ORB-SLAM3 v1.0 integrate with rtabmap in IMU_RGBD mode?

ericlai0323, 2023-10-27:
@matlabbe 
Hello, I'd like to ask if there have been any updates regarding the integration of IMU and RGBD in ORB-SLAM3 with RTAB-Map. Thank you very much!

matlabbe, 2023-10-30:
I created a pull request [there](https://github.com/introlab/rtabmap/pull/1152) till I have more time to debug. There are some issues when enabling IMU... that seems more related to ORB_SLAM3.

matlabbe, 2023-11-19:
I merged this pull request https://github.com/introlab/rtabmap/pull/1152, see pull request description for current issues with ORB-SLAM3 integration. At least for VO (Stereo/RGB-D) mode should work out-of-the-box. For VIO, the crashes seem more related to the ORB-SLAM3 library itself (see comments [here](https://github.com/introlab/rtabmap/pull/1152)).