introlab/rtabmap #887: Support for Orbbec cameras

henriquenunez, 2022-08-12:
Hello, I have a pair of orbbec cameras and I want to try adding support for them. Any tips?

rlabs-oss, 2022-08-21:
without direct support in the rtabmap app, capture the RGB + Depth images from the camera and use them via RGB-D image input specifying the rgb and depth folders that contain the images. An RGB calibration is needed before processing.

henriquenunez, 2022-08-28:
Ok. RGB calibration = camera intrinsics matrix?

rlabs-oss, 2022-08-28:
> 
If you have the SDK for the Orbbec camera it would need integrating similar to the realsense, zed etc stereo cameras which are found under the corelib/src/camera

For experimental work, a stereo pair can be used with source set to stereo and either each left/right pair provided in image folders, or as a video in the case of side by side stereo. - For RGB-D I have used my iPhone, or GoPro and AI depth generation - in that case first recording video from the camera, creating a set of images with ffmpeg, then using the calibrate feature in RTAB-Map to calcualte intrinsics. Followed by setting the source to RGB-D, and then supplying the RGB image set and Depth created by e.g MiDaS or similar for the actual scene. 
Another method that works is to compute a point cloud from the image+depth and use that as a laser scan with ICP matching, no stereo calibration is needed for that.

rlabs-oss, 2022-08-28:
Orbbec Astra is supported via openNI2

henriquenunez, 2022-08-28:
Not for multi camera as far as I've tested.

henriquenunez, 2022-08-28:
Just FIY, I am trying to add support specifically to the Astra S U3 sensor, that falls in a gray zone when it comes to openNI, and their support is a bit weird (even with their own drivers). Take a look
[image]

matlabbe, 2022-10-24:
Orbbec is supported through openni2:
[image]

for multi-cameras, you should use ROS, there is an example there: https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch

I don't own an astra S U3 sensor, so if it is not compatible OpenNI2 by default, it would be difficult to me to make the driver in the standalone. When a driver is not available in the standalone, a workaround is to is the camera under ROS (if they provide a ROS driver) then just connect the right topics to rtabmap node. See multiple examples here: http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping and http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping