introlab/rtabmap #889: 3D Map viewing issues on Jetson Xavier NX

sumitsarkar1, 2022-08-19:
1.) Standalone version of RTAB-Map looks this way. Even loaded .db files are not able to be viewed.
2.) Searched the web on nvidia forums and others...could not figure out much from the solutions....any lead to a solution where this problem is fixed or any suggestions will be really helpful.
[image]

closed 2022-08-22 by sumitsarkar1

pramodwic1, 2022-08-24:
Hi @sumitsarkar1 

I too have been working on installing Rtabmap on Xavier NX (with intel realsense D455). Rtabmap and the 3D map works fine after installing most libraries from source as mentioned here (https://github.com/introlab/rtabmap/issues/427#issuecomment-1000921742).

However I noticed a significant lag while running rtabmap and mapping 3D point cloud, seems like the lag increases with time to the point where it may take 5 seconds for the 3D point cloud to update to new camera position. 

Could I know your experience with it?, is it working fine for your use case?

sumitsarkar1, 2022-08-27:
I am using ZED camera with RTabMap...right now we are experimenting with small spaces...we are just running it for 2 or 3 minutes....let me run it for more time and come back to your querry....but as of now it runs smooth...