introlab/rtabmap #890: run rtabmap segmentation fault

Userpc1010, 2022-08-22:
Platform | Desktop
-- | --
OS | Ubuntu 18.04.6 LTS
Kernel Version | 5.4.0-124-generic
Device: | Desktop AMD FX8320 / Nvidia 750Ti 
CODE | Source Code From Last github

After run rtabmap from source: 

```
sencis@sencis-desktop:~/build-rtabmap-Desktop-Default$ gdb -r ./bin/rtabmap
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/rtabmap...expanding to full symbols...(no debugging symbols found)...done.
(gdb) r
Starting program: /home/sencis/build-rtabmap-Desktop-Default/bin/rtabmap 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffb3e2a700 (LWP 3220)]
[New Thread 0x7fffb3629700 (LWP 3221)]
[New Thread 0x7fffaee28700 (LWP 3222)]
[New Thread 0x7fffac627700 (LWP 3223)]
[New Thread 0x7fffa9e26700 (LWP 3224)]
[New Thread 0x7fffa7625700 (LWP 3225)]
[New Thread 0x7fffa4e24700 (LWP 3226)]
[New Thread 0x7fffa2623700 (LWP 3227)]
[New Thread 0x7fffa1e22700 (LWP 3228)]
[New Thread 0x7fffa1621700 (LWP 3229)]
[New Thread 0x7fffa0e20700 (LWP 3230)]
[New Thread 0x7fffa061f700 (LWP 3231)]
[New Thread 0x7fff9fe1e700 (LWP 3232)]
[New Thread 0x7fff9f61d700 (LWP 3233)]
[New Thread 0x7fff9ee1c700 (LWP 3234)]
[New Thread 0x7fff9b4d9700 (LWP 3235)]
[New Thread 0x7fff9395e700 (LWP 3236)]
[New Thread 0x7fff9315d700 (LWP 3237)]
[New Thread 0x7fff91f0c700 (LWP 3238)]
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x7fff81908700 (LWP 3240)]
[ WARN] (2022-08-22 22:44:48.758) Parameters.cpp:1237::readINI() Section "Core" in /home/sencis/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
[ WARN] (2022-08-22 22:44:50.235) StatsToolBox.cpp:651::setWorkingDirectory() The directory "/home/sencis/Documents/RTAB-Map" doesn't exist, using "/home/sencis" instead...
Program started...
[New Thread 0x7fff80b1d700 (LWP 3241)]

Thread 1 "rtabmap" received signal SIGSEGV, Segmentation fault.
0x00007ffff4f38f80 in _int_malloc () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) 
```
cmake:

```
sencis@sencis-desktop:~/rtabmap/build$ cmake ..
-- The C compiler identification is GNU 6.5.0
-- The CXX compiler identification is GNU 6.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
[... 203 lines cut ...]
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sencis/rtabmap/build
sencis@sencis-desktop:~/rtabmap/build$ 
```

sumitsarkar1, 2022-08-25:
May be the following points , not sure : 

1.) -- Could NOT find Qt5Svg (missing: Qt5Svg_DIR)..install this library ..try sudo apt-get install libqt5svg5
2.) If you build pcl from source check march native flag

matlabbe, 2022-11-06:
Please do `bt` (backtrace) in gdb when the segfault happens (see ). A malloc error seems to be a wrong version library (if multiple versions of the same library exist, like OpenCV) used on runtime. Can you try "-DWITH_GTSAM=OFF"? gtsam seems built from source, make sure it is not built with `-march=native` flag.